open62541/open62541 #2135: XML Nodeset Exception with Opc.Ua.NodeSet2.xml

BigeYoung, 2018-10-23:
Description
===========

**Exception: Reference ns=0;i=15100<--[ns=0;i=38]--ns=0;i=426 has an unknown target.**

Run:

`python ./nodeset_compiler.py --types-array=UA_TYPES --existing ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml ModBus.xml MB_Node`

Got:

`INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml`

`INFO:__main__:Preprocessing ModBus.xml`

`Traceback (most recent call last):`
`  File "./nodeset_compiler.py", line 167, in <module>`

`    ns.sanitize()`

`  File "./nodeset.py", line 125, in sanitize`

`    raise Exception("Reference " + str(ref) + " has an unknown target")`
`Exception: Reference ns=0;i=15100<--[ns=0;i=38]--ns=0;i=426 has an unknown target`

There is **no** node "ns=0;i=15100" in my code.

Background Information / Reproduction Steps
===========================================

* **OPCFoundation/UA-Nodeset** commit 5bbf784  on Mar 17
* **Python** 3.7.1

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):  `e2d4b0b`
 - [x] Other OPC UA SDKs used (client or server): `opcua-modeler-0.5.5` _(for create information models)_
 - [x] Operating system: `Ubuntu 16.04 LTS`
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-10-23:
Hello @BigeYoung ,

the node with i=15100 comes from the nodeset2.xml.

```xml
  <UAObject NodeId="i=15100" BrowseName="Default JSON" SymbolicName="DefaultJson">
    <DisplayName>Default JSON</DisplayName>
    <References>
      <Reference ReferenceType="HasEncoding" IsForward="false">i=426</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
    </References>
  </UAObject>
```

BigeYoung, 2018-10-23:
Sorry, I am still a newbie. I don't understand why it leads to errors, since this node is there. My xml can be open in FreeOpcUa Modeler, with "Node Sets" contains "Opc.Ua.NodeSet2.xml" only.

BigeYoung, 2018-10-23:
I've run
```Bash
python ./nodeset_compiler.py --types-array=UA_TYPES --existing ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml ../../examples/nodeset/server_nodeset.xml myNS
```
and got the same exception. So I think that this might not my fault.

Pro, 2018-10-30:
Your solution is not 100% correct. Opc.Ua.NodeSet2.xml should contain all the nodes, including Part4. But if it is working for you, I just close it here.

closed 2018-10-30 by Pro

GeorgeSellner, 2018-10-30:
I have the same result if I use the OPC Foundation/UA-Nodeset [v1.04](https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04)