open62541/open62541 #2136: Add statistics counters to the stack to make troubleshooting easier

jonasgreen88, 2018-10-23:
Description
===========
Our Ethernet devices often has a set of statistics counters presented on the internal webpage or through some interface of the protocol. When it comes to Ethernet itself there are interface and media counters. When it comes to specific protocols it can be counter about number of clients connected, how many IO connections that are active etc. It can also be counter indicating errors, such as rejected connections.

My question is if you guys have though about adding something similar to the Open62541 stack. Every part of the stack could have its own set of counters (e.g. secure channel layer, session layer, DataChange subscription implementation etc.). It can be both counter indicating good things but also bad things, such as rejected clients.

The counters should be available through some public API so they can be published on e.g. an internal webpage.

If it sounds interesting to add this feature we maybe can discuss how it should be implemented in this issue. We see this as an important feature for troubleshooting and might be implementing it sooner or later. To prepare for that it would be good to agree upon a solution.

I have two initial proposals how to implement this feature:
1. Add statistics counters to each layer/implementation of the stack that have relevant counters to present. For example, counters of the sessions is implemented in the session layer. Each layer/implementation offers a public API to access the counters.
2. Add a statistics implementation of the stack containing all different counters and it offers a public API to access them. The different layers/implementations of the stack have to interface with the statistics implementation to increment/reset the counters.

Please come up with other proposals or ideas about this feature!

Background Information / Reproduction Steps
===========================================
Statistics counters for troubleshooting devices implementing this stack.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-10-23:
Yes. This is an important feature for professional use.

Would the following API be enough?
Then we can keep a statistics structure in the server/client.
And just give a pointer to the structure that is continuously updated in the background.

```c
struct UA_NetworkStatistics {
    size_t currentConnections;
    size_t closedConnections;
};
struct UA_SecureChannelStatistics {};
struct UA_SessionStatistics {};

struct UA_ServerStatistics {
    UA_NetworkStatistics ns;
    UA_SecureChannelStatistics scs;
   UA_SessionStatistics ss;
};

struct UA_ClientStatistics {
    UA_NetworkStatistics cs;
    // ...
};

const UA_ServerStatistics * UA_Server_getStatistics(UA_Server *server);
const UA_ClientStatistics * UA_Client_getStatistics(UA_Server *client);

jonasgreen88, 2018-10-23:
Yes, that's look promising.

closed 2021-02-03 by jonasgreen88