open62541/open62541 #2137: WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail

GeorgeSellner, 2018-10-23:
Description
===========
Trying to run 'nodeset_compiler.py' on 3rd-party provided xml
Show stopper at this point. Not sure what this warning means, but followed error.

Background Information / Reproduction Steps
===========================================
trying to build server based on models provided by EUROMAP [http://www.euromap.org/en/euromap83/](http://www.euromap.org/en/euromap83/) and experiencing problems

python ~/ws_OPC/open62541/tools/nodeset_compiler/nodeset_compiler.py 
--verbose
--internal-headers
--types-array=UA_TYPES
--types-array=UA_TYPES_DI
--types-array=UA_TYPES_EUROMAP83
--existing /home/mmtools/ws_OPC/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
--existing /home/mmtools/ws_OPC/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
--xml /home/mmtools/ws_OPC/euromap-82-2/EUROMAP-83/Opc_Ua.EUROMAP83.NodeSet2.xml
/home/mmtools/ws_OPC/euromap-82-2/euromap/ua_namespace_EUROMAP83 

```
INFO:__main__:Preprocessing (existing) /home/mmtools/ws_OPC/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) /home/mmtools/ws_OPC/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing /home/mmtools/ws_OPC/euromap-82-2/EUROMAP-83/Opc_Ua.EUROMAP83.NodeSet2.xml
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
WARNING:nodes:Value ranks >0 not fully supported. Further steps may fail
Traceback (most recent call last):
  File "/home/mmtools/ws_OPC/open62541/tools/nodeset_compiler/nodeset_compiler.py", line 175, in <module>
    ns.buildEncodingRules()
  File "/home/mmtools/ws_OPC/open62541/tools/nodeset_compiler/nodeset.py", line 270, in buildEncodingRules
    n.buildEncoding(self)
  File "/home/mmtools/ws_OPC/open62541/tools/nodeset_compiler/nodes.py", line 567, in buildEncoding
    raise Exception("Node {} not found in nodeset".format(NodeId(fdtype)))
Exception: Node ns=1;i=3002 not found in nodeset
```

[gen83.sh.txt](https://github.com/open62541/open62541/files/2507857/gen83.sh.txt)
[output.txt](https://github.com/open62541/open62541/files/2507866/output.txt)
[EUROMAP83_Release_1.00.pdf](https://github.com/open62541/open62541/files/2507867/EUROMAP83_Release_1.00.pdf)
[EUROMAP83_Types.xsd.txt](https://github.com/open62541/open62541/files/2507873/EUROMAP83_Types.xsd.txt)
[Opc_Ua.EUROMAP83.1_00.NodeId.csv.txt](https://github.com/open62541/open62541/files/2507874/Opc_Ua.EUROMAP83.1_00.NodeId.csv.txt)
[Opc_Ua.EUROMAP83.1_00.NodeSet2.xml.txt](https://github.com/open62541/open62541/files/2507875/Opc_Ua.EUROMAP83.1_00.NodeSet2.xml.txt)
[Opc_Ua.EUROMAP83.NodeId.csv.txt](https://github.com/open62541/open62541/files/2507877/Opc_Ua.EUROMAP83.NodeId.csv.txt)
[Opc_Ua.EUROMAP83.NodeSet2.xml.txt](https://github.com/open62541/open62541/files/2507879/Opc_Ua.EUROMAP83.NodeSet2.xml.txt)

Checklist
=========

Please provide the following information:

 - [x] open62541 Version (release number or git tag):e2d4b0bf0355dfe8a064889f69307169959059b9
Merge pull request #2085 from open62541/runtime_custom_types
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: 4.15.0-36-generic #39-Ubuntu SMP Mon Sep 24 16:19:09 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

GeorgeSellner, 2018-10-29:
The more troubling problem is that it won't generate, due to this error:

> Traceback (most recent call last):
  File "/home/mmtools/ws_OPC/open62541/tools/nodeset_compiler/nodeset_compiler.py", line 175, in <module>
    ns.buildEncodingRules()
  File "/home/mmtools/ws_OPC/open62541/tools/nodeset_compiler/nodeset.py", line 270, in buildEncodingRules
    n.buildEncoding(self)
  File "/home/mmtools/ws_OPC/open62541/tools/nodeset_compiler/nodes.py", line 567, in buildEncoding
    raise Exception("Node {} not found in nodeset".format(NodeId(fdtype)))
Exception: Node ns=1;i=3002 not found in nodeset

It does not like the xml at all. This xml was created in UA Modeler bu EUROMAP
If I use the code generation in UA Modeler, it is able to generate code that works with their SDK

GeorgeSellner, 2018-10-29:
Doesn't seem like anyone is interested in helping with this :(

Pro, 2018-10-30:
Dear @GeorgeSellner, just closing the issue does help you even less, because we are not checking closed issues.

Also keep in mind that we are an open source community and not a company which provides 24/7 support.
If you require commercial support, please check our Readme, there we have listed a few companies which provide that kind of service for our stack.

Anyways I will try to have a look at your issue. The error comes due to a missing node in the nodeset or a missing dependency.

Creating a nodeset in UaModeler does not mean that it is valid.

stsch83, 2018-11-01:
Is there currently a solution to the problem?

commit 9293907e (2018-10-31): NodesetCompiler: Correctly handle namespace mapping for encoded objects. Fixes #2137
commit eb638386 (2018-10-31): NodesetCompiler: Correctly handle namespace mapping for encoded objects. Fixes #2137
PR #2161 (2018-11-02): NodesetCompiler: Correctly handle namespace mapping for encoded objects.
Fixes #2137
commit e1b11ea2 (2018-11-02): NodesetCompiler: Correctly handle namespace mapping for encoded objects. Fixes #2137
closed 2018-11-02 by Pro