open62541/open62541 #2138: Access SecurityPolicies from access control implementation

jonasgreen88, 2018-10-24:
Description
===========
We have implemented the possibility to allow clients encrypt the password of the UserNameIndentityToken when communicating over a secure channel with SecurityPolicy None. This is needed to pass the CTT if using SecurityPolicy None on the secure channel.

For the moment we access the mbed TLS stack directly in the access control plugin implementation. I think we could use the security policy plugin implementations instead. But I don't find a nice way to access the security policies from the access control plugin. Currently, the security policies are stored on every endpoint (UA_Endpoint). As both the endpoints, but also the UserIdentityTokens may need the security policies, wouldn't it be better to store the security policies on a common location (e.g. UA_ServerConfig?) and then have pointer references from the endpoint and the UserIdentityTokens to point out which security policy they use?

Proposal:
-----------
Store the security policies in the server config:
```
struct UA_ServerConfig { 
//
size_t securityPolicySize;
UA_SecurityPolicy *securityPolicies;
//
}
```
Update the UA_Endpoint struct to reference the security policy (instead of storing it):
```
typedef struct {
    UA_SecurityPolicy *securityPolicy;
    UA_EndpointDescription endpointDescription;
} UA_Endpoint;
```
Create a new type for user tokens (similar to the UA_Endpoint struct) that contains a reference to the security policy that should be used and the UserTokenPolicy:
```
typedef struct {
    UA_SecurityPolicy *securityPolicy;
    UA_UserTokenPolicy userTokenPolicy;
} UA_UserToken;
```
Update the UA_AccessControl structure to store UA_UserToken instances (instead of UA_UserTokenPolicy instances).
```
struct UA_AccessControl {
//
    /* Supported login mechanisms. The server endpoints are created from here. */
    size_t userPoliciesSize;
    UA_UserToken *userTokens;
}
```

What do you guys think about this design?

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jonasgreen88, 2018-10-24:
> Yes. Absolutely.
> 
> So we have a central array for securitypolicies where they can be looked up via their URL.
> 
> Somehow we need control over combinations of channel policy and user token policy. Isn’t it enough to configure the endpoints with policy urls. Like we do now?

Where is this central array located? I can only find that the SecurityPolicies are initialized and stored in every enpoint (UA_Endpoint).

The UserIdentityTokens needs to point out their own SecurityPolicy. For example, if you have an endpoint with SecurityPolicy None, that endpoint can have a set of UsernameIdentityTokens using different SecurityPolicies to encrypt the password. This allows the client to choose a security policy of their choice to be used when encrypting the password.

Also, if a server has multiple endpoints using the same SecurityPolicy but maybe different security modes (Sign or SignAndEncrypt), a SecurityPolicy is initialized multiple times. Once for each endpoint. With the design I presented, every SecurityPolicy is only initialized once, which will save some RAM memory.

jpfr, 2018-10-24:
> Where is this central array located? I can only find that the SecurityPolicies are initialized and stored in every enpoint (UA_Endpoint).

We don't have that right now.
It would be the consequence of your proposal. Which I fully support.

jonasgreen88, 2018-10-24:
> It would be the consequence of your proposal. Which I fully support.

Ah, great! If there is a central array for the security policies in the UA_ServerConfig structure the pointer to the security policy in UA_Endpoint and UA_UserToken should not be needed as long as searching the security policy by the policy url is fast enough. In that case the UA_UserToken structure won't be needed and also the UA_Endpoint shouldn't be needed anymore I guess? The endpoints can be represented by the UA_EndpointDescription type in the UA_ServerConfig structure.

Update of the proposal:
--------------------------
Store the security policies in the server config and represent the endpoints with the UA_EndpointDescription type (remove usage of UA_Endpoint):
```
struct UA_ServerConfig { 
//
size_t securityPolicySize;
UA_SecurityPolicy *securityPolicies;

/* Available endpoints */
size_t endpointsSize;
UA_EndpointDescription *endpoints;
//
}
```

jpfr, 2018-11-27:
Full ACK to the proposal @jonasgreen88 .
Are you still on to this?

jonasgreen88, 2019-01-17:
Solved by #2247

closed 2019-01-17 by jonasgreen88