open62541/open62541 #2140: Handle certificate chains in the OpenSecureChannel and CreateSession requests

jonasgreen88, 2018-10-25:
Description
===========
I'm attending the OPC IOP Workshop this week at Siemens in Nürnberg. With one client (using Eclipse Milo SDK) our device returned BadCertificateInvalid. This error code is only returned at one location in the stack: https://github.com/open62541/open62541/blob/e2d4b0bf0355dfe8a064889f69307169959059b9/src/server/ua_services_session.c#L78

We found out that the client included the complete certificate chain (in this case the device certificate + the CA certificate) in the _clientCertificate_ field of the request. But it did only return the device certificate in the CreateSessionRequest. When the stack compares the _remoteCertificate_ of the secure channel and the _clientCertificate_ of the CreateSessionRequest it fails. See attached Wireshark log.
[certificate-chain-opensecurechannel.zip](https://github.com/open62541/open62541/files/2514498/certificate-chain-opensecurechannel.zip)

This is a valid behavior that I think the stack must support. The OPC UA specification Part 6 (1.04), chapter 6.2.3, says:

> However, applications have the option of including a partial or complete chain whenever they pass a Certificate to a peer during the SecureChannel negotiation and during the CreateSession/ActivateSession handshake. All OPC UA applications shall accept partial or complete chains in any field that contains a DER encoded Certificate.
>
> Chains are stored in a ByteString by simply appending the DER encoded form of the Certificates. The first Certificate shall be the end Certificate followed by its issuer. If the root CA is sent as part of the chain it is last Certificate appended to the ByteString.

So, I have done a small investigation what needs to be fixed. Feel free to comment and extend these results.

Secure channel
-----------------
The _SenderCertificate_ of the Asymmetric algorithm security header can contain either just the device certificate of the client, part of the chain or the complete chain CA->device certificate. In part 6 (1.04), chapter 6.7.2.3, table 42, the description of the _SenderCertificate_ says:

> If the Certificate is signed by a CA, the DER encoded CA Certificate may be appended after the Certificate in the byte array. If the CA Certificate is also signed by another CA this process is repeated until the entire Certificate chain is in the buffer or if MaxSenderCertificateSize limit is reached (the process stops after the last whole Certificate that can be added without exceeding the MaxSenderCertificateSize limit).
Receivers can extract the Certificates from the byte array by using the Certificate size contained in DER header (see X.509 v3).
**Receivers that do not handle Certificate chains shall ignore the extra bytes.**

Note the last sentence of the quote (in bold). I interpret that as it should be enough that the stack stores the device certificate and ignores the other certificates if the client have included a chain. In that case it should be enough to update this copy to first get the device certificate of the chain and write it to the _remoteCertificate_ of the channel.
https://github.com/open62541/open62541/blob/e2d4b0bf0355dfe8a064889f69307169959059b9/src/ua_securechannel.c#L75

Session
--------
Also the CreateSession request can contain a certificate chain. So if the _clientCertificate_ of the request is longer than the _remoteCertificate_ of the channel the stack must check if the remoteCertificate is included in the chain.
https://github.com/open62541/open62541/blob/e2d4b0bf0355dfe8a064889f69307169959059b9/src/server/ua_services_session.c#L76
Another thing I found is about the serverSignature of the CreateSession response. In Part 4 (1.04), chapter 5.6.2.2, table 15, the description of the _serverSignature_ says:
> If the clientCertificate contains a chain, the signature calculation shall be done only with the leaf Certificate. For backward compatibility a Client shall check the signature with the full chain if the check with the leaf Certificate fails.
So, the if the _clientCertificate_ of the request contains a chain, the stack must parse out the leaf certificate and use that to create the _serverSignature_. Currently, the stack uses the complete _clientCertificate_ of the request.
https://github.com/open62541/open62541/blob/e2d4b0bf0355dfe8a064889f69307169959059b9/src/server/ua_services_session.c#L47

So this was a quick investigation made at the workshop when founding this issue. There might be further changes needed to be completely compatible with all clients.

Background Information / Reproduction Steps
===========================================
Connect to a Open62541 server with a client that sends the complete certificate chain in the OpenSecureChannelRequest but only the device certificate in the CreateSessionRequest, or vice versa. If the certificate chain is included in the CreateSessionRequest the serverSignature shouldn't be created according to the specification (not tested).

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): cce2d622390be4dafd10ebde80fc2dbb3abadc78 (0.3 branch)
 - [ ] Other OPC UA SDKs used (client or server): Eclipse Milo
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-11-06:
Are we all covered with #2141?

The heavy lifting is done in securityPolicy->channelModule.compareCertificate.
Can we add the comments to the header file that defines compareCertificate?

So that other SecurityPolicy implementations know what is expected for that function.

jonasgreen88, 2018-11-06:
I agree to that the documentation of `compareCertificate()` should be updated to describe that it is the first certificate of each possible chain that shall be compared.

I don't think this issue is completely solved. We cover the case when the clients send a certificate chain in the OpenSecureChannelRequest. But if the CreateSessionRequest contains a certificate chain the `serverSsignature` will be wrong as it is based on the complete client certificate chain included in the request. The specification says that only the leaf certificate of the client shall be used when generating the `serverSignature`.

As the OpenSecureChannel and the CreateSessionRequest can contain only a leaf certificate or a partial/complete certificate chain I don't see the reason why the stack shall store the certificate in the secure channel context and in the session context as byte strings. The stack cannot do anything with that information really, as it doesn't have the capability to parse the data. Instead I think the security policy shall be the only place to store the certificates in runtime and all interactions between the stack and the certificates shall be made through the security policy API. This will solve the remaining part of this issue as well.

closed 2023-09-18 by jonasgreen88