open62541/open62541 #2156: [Question] UA_Server_setNodeContext, at which point is the memory cleaned?

fabian727, 2018-10-30:
Hi @ all,
I would like to add the NodeContext, like stated in:  #1398 

How can I remove the Context again?
If I shutdown the server, is the memory freed by the stack?

Which information should I get from the commentary?
`/* Careful! The user has to ensure that the destructor callbacks still work. */` 

Thanks,
Fabian

Pro, 2018-10-31:
Hi Fabian,
you have to completely handle the node context on your own. It's basically just a storage for any pointer you may need later.
The stack does not free or touch the content of the node context in any case. So you also need to free it on your own.

That comment is a bit unclear and I'm also not sure what the intention of it is...

It was lastly changed here:
https://github.com/open62541/open62541/commit/394df3be8d6a19781927f0790d9c2a9142a2d6a8#diff-92f30b4d60220dac922f98cccefd8992L594

And added here:
https://github.com/open62541/open62541/commit/2113ec1c9b4a96a17e1a8ee98952a71475911712#diff-92f30b4d60220dac922f98cccefd8992R594

@jpfr could you clarify what is meant by that comment?

jpfr, 2018-10-31:
You can add destructor callbacks to nodes. These are called with the context when the node is destroyed.

There’s a section in the server docs.

closed 2018-11-01 by Pro