open62541/open62541 #2166: Client failed to monitor Boolean variable

jonasgreen88, 2018-11-02:
Description
===========
During the OPC IOP Workshop last week one client failed to add a monitored item to monitor a Boolean variable on our device. Our device returned the status code UA_STATUSCODE_BADFILTERNOTALLOWED. The only function of the stack that returns this status code is setMonitoredItemSettings(). 
https://github.com/open62541/open62541/blob/5544ed14aa2a9b88f1cc4388438d2bbe1c2d4f07/src/server/ua_services_subscription.c#L173
We tracked the issue to the function isDataTypeNumeric(), which only allows clients to monitor the data types within the range of SByte->Double to be monitored it the client specifies a filter in the CreateMonitoredItemRequest. The guy that tested the client at the workshop told me that no other server had the same behavior and we couldn't find anything in the specification that motivates that only these data types should be allowed. So I have a couple of questions regarding this implementation:

1. Does the specification somewhere states that it should only be possible to define a filter to "numeric datatypes" (SByte->Double)?
2. If the client doesn't define a filter in the request the server creates a DataChangeFilter itself. But in this case no checks are performed to verify if the variant is empty and if the data type is numeric. Why this difference?

Background Information / Reproduction Steps
===========================================
Send a CreateMonitoredItemRequest that includes a filter definition to a open62541 server to monitor a Boolean variable. The request will be denied.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): cce2d62
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-11-02:
I guess the reason for this check and the return value is that currently we do not have full filter support.
Basic filter support was added in #1726

The check for isDataTypeNumeric is probably there because for now we only support the DeadBand for numeric values:
https://github.com/open62541/open62541/pull/1726/files#diff-a593a0da50c5187578e2b97817717da4R153

As long as we do not have the corresponding DeadBand filter check, we should return some error code for non supported data types when creating the monitored item with filter. Maybe it makes more sense to change the error to `UA_STATUSCODE_BADMONITOREDITEMFILTERUNSUPPORTED`.

If the client does not define a (DeadBand) filter, the server is creating a basic filter internally which can handle simple data change by comparing the value to the previous one.

jpfr, 2018-11-02:
This sounds as if the problem is not the support for filters per se. But the treatment of bool as un-numeric.
It should be straightforward to add bool as a "numeric type" with values zero and one...

closed 2018-11-06 by jpfr