open62541/open62541 #2167: How to read the textual part of an enum value from UA server

sdsbjn, 2018-11-02:
Description
===========
I try to get the texts out of a serverstatus or any enum ?

First I look for items of interest with known types like INT32. 
How to proceed from that point? 
Any short code would be very helpful.

jpfr, 2018-11-02:
That is a good idea. But open62541 currently has no way to retrieve the enum string at runtime.
We generate C-enums from the XML descriptions. But these compile to int32.
The enum string is simply not baked into the library at the moment.

sdsbjn, 2018-11-02:
I ment how to get EnumString / EnumValue from a client.

UA Expert can browse these names and values.  If I select "ServerState" I get DataType 852. If I go to typedef I can browse into EnumStrings and see all enum texts that I'am looking for.

ccvca, 2018-11-02:
As a client you can have a look at the Values from Root->Types->DataTypes->BaseDataType->Enumeration-><The Type you want>->EnumStrings.
I do not know about the support of LocalizedTextArrays, but thats the official location for a client to retrieve the string-Values for enumerations.  (Maybe the Full NS0-Version of open62541 is needed)

sdsbjn, 2018-11-02:
Thanks I understood, but how do I get them in practice?

1) I walk all my vars and found for example
  NS2|String|Demo.Static.Scalar.ServerState and
  NS3|String|AirConditioner_1.State

2) I read the datatype attr - so far so good, but what to do next?
a) how to check, that this datatype is a enum?
b) how to read the EnumString or EnumValues ?

ccvca, 2018-11-02:
- You get a NodeId from the DataType Attribute
- This NodeId is an enum if any of its parents (browse Inverse "SubtypeOf"-Reference) is Enumeration (ns0;i=29) (browse recursive until you either hit Enumeration [ns0;i=29], in this case you have an Enumeration-Type or BaseDatatype; [ns0;i24] (in this case you do not have an enumeration.) )
- If the DataType is an enum, browse (Forward) and look for the BrowseName [0, "EnumStrings"], this is the value with the strings you are looking for.

sdsbjn, 2018-11-02:
Thank you. I'll try my best.

jpfr, 2018-11-05:
This looks done.
Please reopen if the issue remains.

closed 2018-11-05 by jpfr