open62541/open62541 #2168: Missing null checks for callback functions

coder4803, 2018-11-02:
Description
===========
I am using using UA_Client_call_async -function without passing callback function for it. This causes crash as response is handled and null-check is missing.
File: client/ua_client.c
Function: processAsyncResponse
Line: 340

I also looked around and realized that this problem appears at least with most of the similar cases (async calls);

Background Information / Reproduction Steps
===========================================
Code example:
```
UA_StatusCode retval = UA_Client_call_async(m_client, baseNode, methodNode,
                                            0, NULL, NULL, NULL, NULL);
```
As the response is handled, application crashes.

Checklist
=========
 - [X] open62541 Version (release number or git tag): currently latest master (Log Node Ids on AddNodes consistency checks, 03896e4d3c1e6192ca1ee24377cd53dc5eb63c51)

Pro, 2018-11-02:
Hi @coder4803.
Thanks for the Issue report!
Could you maybe create a PR which adds the Nullchecks?
It would take off a bit of the workload from us.

Otherwise maybe @StalderT could have a look at it?

Thanks!

coder4803, 2018-11-02:
Maby I could create fix for that after the weekend. I will report back if that won't happen.

StalderT, 2018-11-02:
@coder4803 can you check with https://github.com/open62541/open62541/pull/2171 ?

PR #2171 (2018-11-03): [master] Fix Missing null checks for callback functions (#2168)
Fixes #2168
PR #2170 (2018-11-03): [0.3] Fix Missing null checks for callback functions (#2168)
Fixes #2168
closed 2018-11-03 by jpfr