open62541/open62541 #2174: Pubsub Connection

Shimasaa, 2018-11-03:
Description
===========
For running pubsub project, I can not find UA_KeyValuePair definition in project, or UA_ServerConfig has no member pubsubTransportLayers. 

Background Information / Reproduction Steps
===========================================
The code example:
 config->pubsubTransportLayers[0] = UA_PubSubTransportLayerUDPMP();
...
UA_KeyValuePair connectionOptions[3];
connectionOptions[0].key = UA_QUALIFIEDNAME(0, "ttl");
...

Checklist
=========
Please provide the following information:

 -   open62541 Version(release number or git tag):currently latest master 
 - Other OPC UA SDKs used (client or server): tutorial_pubsub_connection.c
 -  Operating system:linux

jpfr, 2018-11-05:
Did you activate PubSub in the cmake options?

Shimasaa, 2018-11-06:
Thank you very much, my problem solved by your question :+1:

closed 2018-11-06 by Pro