open62541/open62541 #2186: Name space # miss-match between generated code and runtime

GeorgeSellner, 2018-11-07:
Description
===========
AT run-time receive error message, which appears to indicate name space expected at run-time not same as what was generated in code.

Background Information / Reproduction Steps
===========================================
1) Generate code from nodeset files:
```
python ~/ws_OPC/open62541/tools/generate_datatypes.py --namespace=3 --type-csv=/home/mmtools/ws_OPC/euromap-82-2/EUROMAP82.2_RC1.00.1/EUROMAP83_Types_1_00_ext82_1_RC1_00_1.csv --type-bsd=/home/mmtools/ws_OPC/euromap-82-2/uamodeler/EUROMAP82.2_RC1.00.1/EUROMAP83_1_00_ext82_2_RC1_00_1.bsd /home/mmtools/ws_OPC/zero-3/generated/ua_types_euromap83

python ~/ws_OPC/open62541/tools/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EUROMAP83 --existing /home/mmtools/opc/UA-Nodeset/Schema/Opc.Ua.NodeSet2.xml --existing /home/mmtools/ws_OPC/UA-Nodeset/DI/Opc.Ua.Di.NodeSet2.xml --xml /home/mmtools/ws_OPC/euromap-82-2/EUROMAP82.2_RC1.00.1/EUROMAP83_1_00_ext82_2_RC1_00_1.xml  /home/mmtools/ws_OPC/zero-3/generated/ua_namespace_euromap83

```
2) Build executable
3) Run, errors
`[2018-11-07 09:13:29.078 (UTC-0500)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Type-checking the variable node ns=3;i=6118 failed with error code BadTypeMismatch`

[ua_types_euromap83_generated_handling.h.txt](https://github.com/open62541/open62541/files/2557702/ua_types_euromap83_generated_handling.h.txt)
[EUROMAP82.2_RC1.00.1.zip](https://github.com/open62541/open62541/files/2557703/EUROMAP82.2_RC1.00.1.zip)
[gen_output.txt](https://github.com/open62541/open62541/files/2557704/gen_output.txt)
[gen83g.sh.txt](https://github.com/open62541/open62541/files/2557705/gen83g.sh.txt)
[run_output.txt](https://github.com/open62541/open62541/files/2557706/run_output.txt)
[ua_namespace_euromap83.c.txt](https://github.com/open62541/open62541/files/2557707/ua_namespace_euromap83.c.txt)
[ua_namespace_euromap83.h.txt](https://github.com/open62541/open62541/files/2557708/ua_namespace_euromap83.h.txt)
[ua_types_euromap83_generated.c.txt](https://github.com/open62541/open62541/files/2557709/ua_types_euromap83_generated.c.txt)
[ua_types_euromap83_generated.h.txt](https://github.com/open62541/open62541/files/2557710/ua_types_euromap83_generated.h.txt)
[ua_types_euromap83_generated_encoding_binary.h.txt](https://github.com/open62541/open62541/files/2557711/ua_types_euromap83_generated_encoding_binary.h.txt)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): [f3cb66](f3cb66a68a388fc143c3199015019265f2151c2a)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: 4.15.0-38-generic #41-Ubuntu SMP Wed Oct 10 10:59:38 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-11-08:
Unfortunately the generate_datatypes script is not intelligent enough to automatically handle namespace indices correctly. Currently you need to make sure that you are passing the correct namespace index for the datatypes so that it matches the final namespace index in the server.

See also the docs here: https://github.com/open62541/open62541/blob/master/doc/nodeset_compiler.rst

Are you sure that the 83 nodeset is at namespace index 3 when the server is created?

Also, make sure that your are using current master, since I recently pushed some fixes especially for the Euromap nodeset generation.

GeorgeSellner, 2018-11-08:
@Pro 

Thank you for your help. Currently, I'm not sure of anything at the moment. :)

When I generate name_space for DI, EUROMAP83 and EUROMAP82_2 I set --namespace= to 2,3,4 respectively. If I look in the generated ua_namespace_*.c files, ns= is 1,2,3 respectively. When I call ua_namespace_*() in my main.c I do it in the same order.
I added
`    myNS = UA_Server_addNamespace(server, "http://opcfoundation.org/UA/");
    printf("NS0 Nodeset Namespace = %d\n", myNS);`
and
`    myNS = UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/");
    printf("Added DI Nodeset Namespace = %d\n", myNS);`

NS0 ns=0
DI ns=2

I would assume the two EUROMAP nodesets would be at 3 & 4, but I still get the runtime errors regarding 83 nodes. Still confused. Does the 'BadTypeMismatch' mean this has nothing to do with the namespace, and is another issue due to the peculiarity of EUROMAP?
[output_20181108.txt](https://github.com/open62541/open62541/files/2562270/output_20181108.txt)

GeorgeSellner, 2018-11-09:
Tried again after pull of master.
Also updated UA_Nodesets from OPC Foundation
Still having a issue with certain nodes in the EUROMAP 83

[gen_out_20181109.txt](https://github.com/open62541/open62541/files/2567440/gen_out_20181109.txt)
[run_out_20181109.txt](https://github.com/open62541/open62541/files/2567441/run_out_20181109.txt)

[generate.sh.txt](https://github.com/open62541/open62541/files/2567448/generate.sh.txt)

GeorgeSellner, 2018-11-13:
I was able to get further along by removing nodes in EUROMAP that reference EUInformation and TimeZone offset among others that don't seem to generate properly.
However I have some questions:
The attached screenshot from the OPC client illustrates one of the things I am confused about. Am I doing something wrong that I'm getting 'urn:unconfigured:application' in namespace index 1? I would have expected the DI namespace to have gone in that spot?

[image]

[server_nodeset_euromap.c.txt](https://github.com/open62541/open62541/files/2577757/server_nodeset_euromap.c.txt)

jpfr, 2018-11-18:
The namespace 1 is reserved for the UA application that runs on the server.
So it has the URI that is configured in the ApplicationDescription of the server.

jpfr, 2018-11-18:
With respect to the namespace index of generated datatypes:

There is now a public API for adding datatypes to the server at runtime.
https://github.com/open62541/open62541/blob/master/include/ua_server_config.h#L80

You could manually edit the datatype descriptions once the server is running and the namespace order is determined. And then add the datatype descriptions to the linked list in the server.

GeorgeSellner, 2020-10-05:
The confusion I was having was caused by how UAModeler handles namespaces vs. how they actually work at runtime. UAModeler does not leave a place-holder in the namespace list for the runtime namespace of the server( ns = 1) it just assigns namespace indexes as namespaces are added to the project. So, issue lies with UAModeler and not Open62541.

closed 2020-10-05 by GeorgeSellner