open62541/open62541 #2189: Client: foreign server responds to PublishRequest with BadTimeout and ServiceFault

jrcoding, 2018-11-08:
Description
===========
I tried to monitor items served by an opcua gateway from Softing (uaGate) and got **PublishResponses** with error code **BadTimeout** and **ServerFault.** Monitoring the same item with uaexpert works fine.

Background Information / Reproduction Steps
===========================================
After some debugging I found a differenz between the PublishRequest package send by uaexpert and open62541 that seems to provoke the error. 
Uaexpert sets the **timeoutHint** in the RequestHeader of a **PublishRequest** to 60000 and open62541 sets it to zero.
As a test I've changed it to 60000 in **UA_Client_Subscriptions_backgroundPublish** (see attached patch) and it works so far but I'm not sure if this i so.k. and about other implications this may have.

[PATCH-ua_client_subscriptions.c.gz](https://github.com/open62541/open62541/files/2561760/PATCH-ua_client_subscriptions.c.gz)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):  _0.4_
 - [x] Other OPC UA SDKs used (client or server): _ua-sdk_ _from unified automation_
 - [x] Operating system: _Linux (Debian 10)_
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2018-11-12:
Yes. We can add the timeouthint as you suggested.
Do you want to prepare a pull request? Otherwise we can take your patch and go from there.

jrcoding, 2018-11-13:
First of all, great work the open62541 project!!!

Please feel free using the attached patch....
(but if wanted I can also provide a pull request, your choice)

BTW: 
Does it make sense to have such timeouts configurable, for example as items in the client config???

jrcoding, 2018-11-16:
thanks for merging!

closed 2018-11-16 by jrcoding