open62541/open62541 #2199: chunk desynchronization of the processing of incoming messages

StalderT, 2018-11-09:
When several messages arrive at the same time, sometimes the process becomes out of sync and the messages are not processed in the order of arrival. This causes the connection to close because the sequence number of the messages is no longer correct : 

https://github.com/open62541/open62541/blob/28081f7c2475627b429afbdc3c940d7afbfe25e3/src/ua_securechannel.c#L962

Steps to reproduce the issue:

Start ```bin/examples/server_ctt``` from 0.3
Start ```bin/examples/client_subscription_loop``` from https://github.com/open62541/open62541/pull/2198

The problem sometimes happens quickly sometimes it is necessary to wait longer.

Results :
=========

```
RECEIVE 280 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 5d 00 00 00 5d 00 00 00 01 00 3d 03 c0 6f 2d 55 2c 78 d4 01 5d 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 02 00 00 00 01 00 00 00 1c 00 00 00 00 1c 00 00 00 c0 6f 2d 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 02 00 00 00 0d 0d f0 13 09 55 2c 78 d4 01 f0 13 09 55 2c 78 d4 01 18 14 09 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 5e 00 00 00 5e 00 00 00 01 00 3d 03 7c 72 2d 55 2c 78 d4 01 5e 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 01 00 00 00 01 00 00 00 1c 00 00 00 00 1c 00 00 00 7c 72 2d 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 01 00 00 00 0d 0d 68 14 09 55 2c 78 d4 01 68 14 09 55 2c 78 d4 01 90 14 09 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=93 AND REQID=93
PROCESS SYM SEQ 93
##################################################### I DECRYPT MESSAGE WITH SEQ=94 AND REQID=94
PROCESS SYM SEQ 94
RECEIVE 420 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 5f 00 00 00 5f 00 00 00 01 00 3d 03 2a af 79 55 2c 78 d4 01 5f 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 03 00 00 00 01 00 00 00 1d 00 00 00 00 1d 00 00 00 2a af 79 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 03 00 00 00 0d 0d ea 68 55 55 2c 78 d4 01 ea 68 55 55 2c 78 d4 01 3a 69 55 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 60 00 00 00 60 00 00 00 01 00 3d 03 d8 b5 79 55 2c 78 d4 01 60 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 02 00 00 00 01 00 00 00 1d 00 00 00 00 1d 00 00 00 d8 b5 79 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 02 00 00 00 0d 0d 0c 6a 55 55 2c 78 d4 01 0c 6a 55 55 2c 78 d4 01 34 6a 55 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 61 00 00 00 61 00 00 00 01 00 3d 03 0c b9 79 55 2c 78 d4 01 61 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 01 00 00 00 01 00 00 00 1d 00 00 00 00 1d 00 00 00 0c b9 79 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 01 00 00 00 0d 0d 70 6a 55 55 2c 78 d4 01 70 6a 55 55 2c 78 d4 01 8e 6a 55 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=95 AND REQID=95
PROCESS SYM SEQ 95
##################################################### I DECRYPT MESSAGE WITH SEQ=96 AND REQID=96
PROCESS SYM SEQ 96
##################################################### I DECRYPT MESSAGE WITH SEQ=97 AND REQID=97
PROCESS SYM SEQ 97
RECEIVE 280 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 62 00 00 00 62 00 00 00 01 00 3d 03 02 e5 c5 55 2c 78 d4 01 62 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 03 00 00 00 01 00 00 00 1e 00 00 00 00 1e 00 00 00 02 e5 c5 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 03 00 00 00 0d 0d 3c 9d a1 55 2c 78 d4 01 3c 9d a1 55 2c 78 d4 01 82 9d a1 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 63 00 00 00 63 00 00 00 01 00 3d 03 6c e9 c5 55 2c 78 d4 01 63 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 02 00 00 00 01 00 00 00 1e 00 00 00 00 1e 00 00 00 6c e9 c5 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 02 00 00 00 0d 0d 72 9e a1 55 2c 78 d4 01 72 9e a1 55 2c 78 d4 01 90 9e a1 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=98 AND REQID=98
PROCESS SYM SEQ 98
RECEIVE 140 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 64 00 00 00 64 00 00 00 01 00 3d 03 c4 eb c5 55 2c 78 d4 01 64 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 01 00 00 00 01 00 00 00 1e 00 00 00 00 1e 00 00 00 c4 eb c5 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 01 00 00 00 0d 0d d6 9e a1 55 2c 78 d4 01 d6 9e a1 55 2c 78 d4 01 fe 9e a1 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=100 AND REQID=100
PROCESS SYM SEQ 100
############### BUG : WAITING 99 BUT RECEIVE 100
RECEIVE 135 BYTES 4f 50 4e 46 87 00 00 00 01 00 00 00 2f 00 00 00 68 74 74 70 3a 2f 2f 6f 70 63 66 6f 75 6e 64 61 74 69 6f 6e 2e 6f 72 67 2f 55 41 2f 53 65 63 75 72 69 74 79 50 6f 6c 69 63 79 23 4e 6f 6e 65 ff ff ff ff ff ff ff ff 65 00 00 00 6c 00 00 00 01 00 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 00 00 00 00 01 00 00 00 02 00 00 00 30 6b c6 55 2c 78 d4 01 20 4e 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=101 AND REQID=108
PROCESS ASYM SEQ 101
##################################################### I DECRYPT MESSAGE WITH SEQ=99 AND REQID=99
PROCESS SYM SEQ 99
############### BUG : WAITING 102 BUT RECEIVE 99
RECEIVE 140 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 02 00 00 00 66 00 00 00 65 00 00 00 01 00 3d 03 ba 3f 12 56 2c 78 d4 01 65 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 03 00 00 00 01 00 00 00 1f 00 00 00 00 1f 00 00 00 ba 3f 12 56 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 03 00 00 00 0d 0d 96 dd ed 55 2c 78 d4 01 96 dd ed 55 2c 78 d4 01 f0 dd ed 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=102 AND REQID=101
PROCESS SYM SEQ 102
RECEIVE 280 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 02 00 00 00 67 00 00 00 66 00 00 00 01 00 3d 03 fc 48 12 56 2c 78 d4 01 66 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 02 00 00 00 01 00 00 00 1f 00 00 00 00 1f 00 00 00 fc 48 12 56 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 02 00 00 00 0d 0d 08 df ed 55 2c 78 d4 01 08 df ed 55 2c 78 d4 01 30 df ed 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff 4d 53 47 46 8c 00 00 00 01 00 00 00 02 00 00 00 68 00 00 00 67 00 00 00 01 00 3d 03 a6 4e 12 56 2c 78 d4 01 67 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 01 00 00 00 01 00 00 00 1f 00 00 00 00 1f 00 00 00 a6 4e 12 56 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 01 00 00 00 0d 0d 94 df ed 55 2c 78 d4 01 94 df ed 55 2c 78 d4 01 bc df ed 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=103 AND REQID=102
PROCESS SYM SEQ 103
##################################################### I DECRYPT MESSAGE WITH SEQ=104 AND REQID=103
PROCESS SYM SEQ 104
RECEIVE 140 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 02 00 00 00 69 00 00 00 68 00 00 00 01 00 3d 03 dc 94 5e 56 2c 78 d4 01 68 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 03 00 00 00 01 00 00 00 20 00 00 00 00 20 00 00 00 dc 94 5e 56 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 03 00 00 00 0d 0d c0 3e 3a 56 2c 78 d4 01 c0 3e 3a 56 2c 78 d4 01 06 3f 3a 56 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff
##################################################### I DECRYPT MESSAGE WITH SEQ=105 AND REQID=104
PROCESS SYM SEQ 105
```

Explanations
=========

The client receive 2 chunks messages : 

```
RECEIVE 280 BYTES 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 62 00 00 00 62 00 00 00 01 00 3d 03 02 e5 c5 55 2c 78 d4 01 62 00 00 00 00 00 00 00 00 ff ff ff ff 00 00 00 03 00 00 00 01 00 00 00 1e 00 00 00 00 1e 00 00 00 02 e5 c5 55 2c 78 d4 01 01 00 00 00 01 00 2b 03 01 26 00 00 00 01 00 00 00 03 00 00 00 0d 0d 3c 9d a1 55 2c 78 d4 01 3c 9d a1 55 2c 78 d4 01 82 9d a1 55 2c 78 d4 01 ff ff ff ff 01 00 00 00 00 00 00 00 ff ff ff ff 4d 53 47 46 8c 00 00 00 01 00 00 00 01 00 00 00 63 00 00 00 63 00 00 00 01 00 3d 03 6c e9 c5 55 2c 78 d4 01 63 00 00 00 [... cut ...]
[image]

The second contains 1 message (seq 100): 

[image]
 
But the client does not process the messages in the order of arrival, process message 98, 100, 101 and then 99 : 

[image]

@jpfr can you take a look ?

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0.3
 - [X] Self-contained code example attached
 - [X] Critical issue

https://github.com/open62541/open62541/issues/1844 https://github.com/open62541/open62541/issues/853

jpfr, 2018-11-09:
This sounds similar to #1930 with the fix in
#1956
#2012
#1940 

There, the problem was the client sending out requests (and processing the new response) when some chunks in the old response had not been processed.

We assumed that this was related only to the async client.
Because that was the only part that needed fixing.
Apparently not.

StalderT, 2018-11-09:
> We assumed that this was related only to the async client.

Background subscription use async.

@jpfr the problem always occure near a openSecureChannel response. In the example sequence number 101 is a opensecurechannel response : 

[image]

StalderT, 2018-11-09:
> Maybe we can ask the client to only renew the SecureChannel when all current chunks have been processed.

I think the problem is not when we send the OpenSecureChannel request but when we receive the OpenSecureChannel response.

I'm trying to understand why we do not process messages in the order of reception. If you can help me :-)

jpfr, 2018-11-09:
We call the networklayer to receive the OpenSecureChannel response.
But there are chunks still in the buffer (returned from the networklayer the last time).

StalderT, 2018-11-10:
@jpfr I also want to merge https://github.com/open62541/open62541/pull/2012/commits/5e73fdbda01961bd87ac292e85c0c3176ea31c09 into 0.3.

I just have a question. Can not there be a problem here if a message has a size bigger than connection-> config.recvBufferSize?

https://github.com/open62541/open62541/blob/e9445c401728ff4e7af17f13d8e9e2e4a779b168/arch/ua_network_tcp.c#L121-L125

jpfr, 2018-11-11:
The message is never bigger than connection->config.recvBufferSize.
This is what we allocate.

And the maximum size is forwarded to the `recv` call.

StalderT, 2018-11-11:
> The message is never bigger than connection->config.recvBufferSize.

If we have a small recvBufferSize we can't have a big message (with many monitoreditemresonse by example) ?

jpfr, 2018-11-11:
Closed via #2202.

PR #2202 (2018-11-11): prevent reentrant renew securechannel
prevent reentrant renew secure channel

fix #2199 #1844
closed 2018-11-11 by jpfr