open62541/open62541 #2201: OPC UA Server with user name and password

TechAspirant, 2018-11-09:
Description
===========
I am trying to connect OPC UA server with user name and password in linux. i have downloaed open62541.c and open62541.h file from https://open62541.org/. My server without username and password works well with OPC client. But how can I give user name and password to my server so that client can connect wit it with credentials.
I cannot see enableAnonymousLogin flag in .c and .h file but I have UA_AccessControl_default in my files but don't know how to use this with my server.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2018-11-09:
I am not sure, but maybe you search for the following lines:
[image]

Within the function createDefaultConfig(void) the function  UA_AccessControl_default(...) will be called and the usernamepassword policy will be enabled.

TechAspirant, 2018-11-10:
Thanks it worked.

closed 2018-11-10 by TechAspirant