open62541/open62541 #2205: What to pass for trustList & trustListSize in server ?

TechAspirant, 2018-11-10:
Description
===========
I am using UA_ServerConfig_new_basic128rsa15 in server , I have certificate file and its key file but don't know what to pass for Trustlist and its size. I am referring below code butin that case what is argc > 3 ?
```
size_t trustListSize = 0;
    if(argc > 3)
        trustListSize = (size_t)argc-3;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
    for(size_t i = 0; i < trustListSize; i++)
        trustList[i] = loadFile(argv[i+3]);
```

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-11-12:
argc is the number of command line arguments.
From the mbedtls documentation, trust lists are .pem files with the certificates you trust.

SebsSa, 2019-06-07:
If it is possible, I would like to take up the subject again.

I'm at the same point as the creator of this topic, my server is running and I've got certificate file and its key file (create_self-signed.py)

If the TrustList is empty the server will accept every "Client Certificate" but my plan would be to only allow access via my locally running UAExpert.
So I opened UAExpert grabbed the "Own Certificate".
(Settings -> Manage Cert.. -> Click "Own Cert.." -> Copy Application Cert.. to ")
Used OpenSSL to convert it.

` openssl x509 -inform der -in ua.cer -out ua.pem`

Now the question how do I link this .pem file ?

`./serverTLS server_cert.der server_key.der ua.pem`

Nothing's happening, I'm pretty sure I have a simple syntax error.
Could somebody help me out?

closed 2020-07-07 by jpfr