open62541/open62541 #2209: undefined reference to `UA_DateTime_now' FreertosLWIP

embbo, 2018-11-12:


jpfr, 2018-11-12:
Maybe your architecture is not baked into the amalgamated files.
Can you use the "normal" source files?

Which architecture do you use?
It should come with a clock implementation.

https://github.com/open62541/open62541/blob/master/arch/freertosLWIP/ua_clock.c
https://github.com/open62541/open62541/blob/master/arch/win32/ua_clock.c
https://github.com/open62541/open62541/blob/master/arch/posix/ua_clock.c
https://github.com/open62541/open62541/blob/master/arch/vxworks/ua_clock.c
https://github.com/open62541/open62541/blob/master/arch/wec7/ua_clock.c

embbo, 2018-11-15:
it backs ok , it includes ua_clock part in open62541.c and also declare in ua_types bt stills gives error .
what can be the issue:
my cmake command is :
/cmake.exe .. -G "Eclipse CDT4 - Unix Makefiles" -DUA_ARCHITECTURE="freertosLWIP"

jpfr, 2018-11-16:
Is this file compiled?
https://github.com/open62541/open62541/blob/master/arch/freertosLWIP/ua_clock.c

Maybe you can regenerate the build system from scratch.

embbo, 2018-11-21:
it works fine , but now its doesn't connct to uaExpert client, i can ping and can see on Wireshark

Pro, 2018-11-22:
Can you post additional information how you fixed this? It may be helpful for others if they find this issue.

For your second question, please open a new issue and provide more info.

embbo, 2018-11-27:
okay, I just used this cmake comand and include freertosLWIP arch in it.
cmake.exe .. -G "Eclipse CDT4 - Unix Makefiles" -DUA_ARCHITECTURE="freertosLWIP"

undefined these two function below  while cmake. (there can be other ways) and define manually when while using with attolic or eclipse. then i complied and  it works really fine and i can see on wireshark.

//UA_Int64 UA_DateTime_localTimeUtcOffset(void) 
//UA_DateTime UA_DateTime_nowMonotonic(void)

embbo, 2018-11-27:
Here is second issue, i have no idea why i can connect . Stm_freertos_LWIP #2227

cabralfortiss, 2019-02-17:
I think this issue could be closed now

closed 2019-02-25 by Pro