open62541/open62541 #2210: Fails to authenticate session

kjeldflarup, 2018-11-13:
Description
===========
I cannot get open62541 to authenticate.

Background Information / Reproduction Steps
===========================================
I'm trying to put a username and password on my service.

I first try to disallow anonymous login in the server.
```
    conf->accessControl = UA_AccessControl_default(/* allowAnonymous */ false, 
                    serverusernamePasswordsSize, serverusernamePasswords);
```
But it is still possible to connect and browse all values as anonymous.

Next I try to set some username and passwords
```
   const size_t serverusernamePasswordsSize = 2;
    UA_UsernamePasswordLogin serverusernamePasswords[2] = {
        {UA_STRING_STATIC("root"), UA_STRING_STATIC("secret")},
        {UA_STRING_STATIC("user2"), UA_STRING_STATIC("password1")}};
```
But it is still possible to browse, specifying both anonymous and any username.

I have a suspicion that I should write an activateSession function, but I can find no documentation on that. Yet still I would have expected that at least setting allowAnonymous = false would reject an anonymous session before getting to that.
```
conf->accessControl.activateSession = activateSession;
```

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
```
commit e9445c401728ff4e7af17f13d8e9e2e4a779b168
Date:   Thu Nov 8 14:46:20 2018 +0100
```
`cmake -DUA_ENABLE_ENCRYPTION=true -DUA_ENABLE_IMMUTABLE_NODES=true -DUA_ENABLE_AMALGAMATION=true .`
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
Ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
[anon.zip](https://github.com/open62541/open62541/files/2575706/anon.zip)

 - [ ] Critical issue

jpfr, 2018-11-13:
Can you point your debugger to this location?
https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c#L38

There is little room for error in this check...

The check is done in your example's open62541.c, line 47088.

kjeldflarup, 2018-11-14:
Now I got it. I have overridden activateSession, but when doing that all the checks made in activateSession_default must also be done. 
Or activateSession_default chained from my activateSession.

Thanks.

closed 2018-11-14 by kjeldflarup

kjeldflarup, 2018-11-14:
A correction, it is not possible to write my own `activateSession`. `struct AccessControlContext` is only declared in `open62541.c`  
Neither is it possible to chain to `activateSession_default` as this function is also just defined in `open62541.c`