open62541/open62541 #2212: MDNS: Handle multiple A-records

ccvca, 2018-11-13:
Description
===========
I discovered some problems on windows with multiple network adapters and mdns-discovery

- Mdns finds multiple IP-Adresses on my computer (one for each networkinterface, **even if the adapter is disabled.**)
- When receiving the record only the first A-record seems to be used,

I would prefer two fixes:
- Do not add IP-adresses from disconnected network adapters, as the 169.254.*.* adresses could not be connected by anyone. (Not even on the same machine)
- Either call the ServerOnNetworkCallback-function multiple times (per IP once), or change the datastructure so multiple URLs could be used

Background Information / Reproduction Steps
===========================================
The captured mdns-package, (the 141.-Adress is the correct external address of the network adapter, all other adapers are either disabled or disconnected)

[image]

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 5a06ca2471fa973cddbac90553cd516db0809f04 with #2211 merged
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached (availiable on request by mail), if the picture is not enought
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2018-11-27:
I've just think about another solution:
What about sending an individual mDNS-Message per Interface, which contains only the IP-Address of that interface. As mDNS does only works inside an subnet, other IP-Adresses might not be routable between the different connected Networks, especially when private IP-Adresses are used (10.0.0.0/8 192.168.0.0/16, 172.16.0.0/12)

Pro, 2018-11-27:
> What about sending an individual mDNS-Message per Interface, which contains only the IP-Address of that interface.

To me that makes perfectly sense, but it may not be so easy to implement since the mDNS library is handling some of the stuff separately. You can try to create a PR and then we can check the solution. That would be really nice :+1:

Pro, 2019-01-04:
This will be solved with #2344. We should not use the A record at all since the spec defines that the SRV announce should always include the hostname, and not an IP address. The OS will then resolve the hostname to the corresponding IP.

commit 1bf487ab (2019-01-04): mDNS: Remove resolving IP address for mDNS announce

Fixes #2322
Fixes #2212
closed 2019-01-04 by Pro