open62541/open62541 #2214: Access to x509parse_keyfile

kjeldflarup, 2018-11-14:
I need to read a .pem file with a certificate. However open62541 expects to get some binary strings.   
I do not want to convert the .pem file to .der externally, I already have management of .pem files on my system.

Basically a base64 decoder could do it, but why add and manage such code, when mbed tls offers x509parse_keyfile which does the job. 

Also in the examples a loadFile is implemented and called, but why use that when mbed tls offers all kinds of file parsing.

ChHaa, 2018-11-14:
The OPC UA standard at some points can be read in a way that DER encoded certificates are the way to store certificates locally.
Only private keys are expected to be .pem encoded, e.g. see ANNEX in part 6

schroeder-, 2018-11-21:
If you want to read a PEM file just add a '\0' at the end of the string.

jpfr, 2019-02-15:
Seems to be resolved.
Please reopen if required.

closed 2019-02-15 by jpfr

HuberHans, 2019-08-15:
Example client_connect with encryption was only working with .der file but not with .pem file.
As stated above adding \0 solves it (which took me 1h to find out by myself).
mbedtls_pk_load_file adds automatically \0 with following code (which is not done in used loadFile function):
    if( strstr( (const char *) *buf, "-----BEGIN " ) != NULL )
        ++*n;
I would like to support both formats like mbed does.
Should the loadFile function be part of SecurityPolicy plugin (I don't know if other libs like openssl have special code/check in loadFile function)?
Is there another general solution I can use?