open62541/open62541 #2218: New Feature: Add AppVeyor build with VS2017

escherstair, 2018-11-15:
AppVeyor is configured to build with VS2008, VS2013 and VS2015.
Since AppVeyor provide a pre-configured image with VS2017 too, it's quite easy to add a new build with VS2017.

If this seems a good idea I can propose a PR.

Pro, 2018-11-22:
Is there a specific advantage to add VS2017?

Currently AppVeyor is already slow and we often have to wait hours until CI runs through. Adding an additional build will slow that down even more.

Pro, 2018-11-23:
@escherstair it would be nice if you can propose a PR which includes the versions mentioned by @jpfr 
(and of course including the other special versions like mingw and clang)

escherstair, 2018-11-23:
The reason to test with VS2017 is to use the latest code analysis tools.

@Pro sure I'm going to propose a PR..
To be 100% sure: in other words, the proposal from @jpfr means to replace VS2015 with VS2017.
Can you confirm?

jpfr, 2018-11-23:
> To be 100% sure: in other words, the proposal from @jpfr means to replace VS2015 with VS2017.
Can you confirm?

Yes. Maybe wen can have both for 1,2 weeks and then disable VS2015.

escherstair, 2018-11-28:
PR #2248  adds build for vs2017 side by side to vs2015.
If everything is ok we can leave it for 1 or 2 weeks, then disable vs2015.

Pro, 2018-11-28:
I just merged the PR.

Let's leave this issue open until around December 15th 2018, then remove VS2015 and close this.

Pro, 2018-12-05:
Closing. Removed in #2274

closed 2018-12-05 by Pro