open62541/open62541 #2221: Cleanup AppVeyor configuration

escherstair, 2018-11-15:
While trying to add clang build to AppVeyor #2116 I saw that AppVeyor configuration is a little bit complicated (I mean, more complicated than necessary with up-to-date AppVeyor infrastructure).
Probably the configuration files have been created some years ago, when AppVeyor was not so full-featured as it is now.

If someone is interested I can cleanup a little bit the configuration files.
This wuold take advantage of [software pre-installed on Windows build VMs](https://www.appveyor.com/docs/windows-images-software/) and this would save build time.

Moreover the AppVeyor configuration files would be a little bit easier to mantain.

If you think it would be a good idea I can propose a PR.

jpfr, 2018-11-15:
Yes, please. 👍 
Simplification is always welcome.

escherstair, 2018-11-27:
I've just created PR #2242

Pro, 2018-11-29:
Done with PR

closed 2018-11-29 by Pro