open62541/open62541 #2227: Stm_freertos_LWIP

embbo, 2018-11-17:
Description
===========

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

FishEatingCat, 2019-01-02:
Hello there!
I also encountered a similar problem, lwip does not add open62541, lwip runs perfectly, but after the open62541 compiled, the network can not ping, through single-step debugging, I found that the problem is in the open62541 call lwip scan_select () Function, I don't know why open62541 calls the underlying code of lwip, why the network can't connect, I don't know what your problem is?

Pro, 2019-01-02:
@cabralfortiss ping

embbo, 2019-01-07:
@FishEatingCat  It complies fine , start brodcasting, but i think the problem is  about Network layer (may be assinging IP )

FishEatingCat, 2019-01-09:
> @FishEatingCat It complies fine , start brodcasting, but i think the problem is about Network layer (may be assinging IP )

I am using a static IP. I mean, my code is compiled without problems. After adding open62541, if you run the server, that is, call UA_Server_run(), then the network will not work; if you comment out the line code, the network is connected.

cabralfortiss, 2019-01-14:
can you both explain what exactly did work and what did you add so it stopped working?

Do you run open62541 in its own task? Do you initialize LwIP before calling anything on open62541?

FishEatingCat, 2019-01-15:
> can you both explain what exactly did work and what did you add so it stopped working?
>My workflow:
1, STM32 underlying hardware driver is no problem
2, transplant freertos system
3, transplant lwip
4, transplant open62541 

Before I added open62541, the first three steps can run normally. lwip communicates dynamically whether it is IP or static IP.

> Do you run open62541 in its own task? Do you initialize LwIP before calling anything on open62541?
> The code of open62541 is in a separate task function. The lwip initialization is completed before the start of the freertos task scheduling, and it works fine.

> did you remove the timers somehow? (I saw the other bug)
The function that gets the current time inside open62541 internally calls the time function of the operating system, but the microcontroller or freertos does not have this type of function, so I replaced it with the RTC time of the microcontroller.

The problems currently encountered:
The previous stage: After adding the open62541 task, the other tasks could not be called. After troubleshooting, it was found to be freertos, and then solved.
At this stage: the open62541 task and the rest of the tasks can be run separately, but the network cannot be connected.

Conclusion: The UA_Server_run() function is not called inside the open62541 task. If the opc ua service is not enabled, the network can communicate normally with the internal and external networks.
The UA_Server_run() function is called, that is, after the opc ua service is enabled, the network cannot communicate with the internal and external networks.

cabralfortiss, 2019-01-15:
Thanks for detailed information. I'm running out of ideas...

How do you set the IP for the opcUa server? are you setting it the same as the one in the microcontroller? 

Only the network is not working or other tasks are also failing? 

The next step I would say is to debug step by step UA_Server_run() and see until where it reaches. I would expect some time error or inifinte loop somewhere that's not properly handled.

FishEatingCat, 2019-01-16:
Thank you very much for your patience and timely reply!
The opc ua service IP address is the same as the IP of the network, which I am sure because I am using a static IP address.

I am very sorry. After replying to your message yesterday, I checked my project and the operation again. I replied that the network does not work after the opc ua service is enabled. The real situation is that after the opc ua service is started, the other network is the same. The segment's computer is able to receive a ping reply. I am really sorry, I have given you the wrong information.

This is the information printed by my microprocessor serial port.
[image]

This is the prompt when ua_export connects to my opcu ua server.
[image]

This is my laptop pinging my microprocessor IP
[image]

I am really sorry

FishEatingCat, 2019-01-17:
After the query, the Header checsum is not sent in the first TCP handshake message initiated by the client. In the second TCP handshake, the server returns the RST, rejecting the client's request, and the server returns a header with a header. Checsum, I think this will be a reason?

FishEatingCat, 2019-01-17:
Oh, I almost forgot, I opened the debugging switch of lwip, then it printed the following content
```
Dno LCD ID:5510
网卡en的MAC地址为:................2.0.0.47.0.37
静态IP地址........................192.168.31.231
子网掩码..........................255.255.255.0
默认网关..........................192.168.31.1

lwip_socket(PF_INET, SOCK_STREAM, 6) = 0
lwip_setsockopt(0, SOL_SOCKET, optname=0x4, ..) -> on
lwip_ioctl(0, FIONBIO, 1)
lwip_bind(0, addr= 127.   0.   0.   1 port=4840)
lwip_bind(0) succeeded
lwip_listen(0, backlog=100)
[2019-01-17 14:59:02] info/network	TCP network layer listening on opc.tcp://192.168.31.231:4840/
lwip_select(1, 680127b8, 00000000, 680127bc, tvsec=       0 tvusec=   50000)
lwip_selscan: fd=0 not ready for reading
lwip_selscan: fd=0 not ready for writing
lwip_selscan: fd=0 not ready for exception
lwip_selscan: fd=0 not ready for reading
lwip_selscan: fd=0 not ready for writing
lwip_selscan: fd=0 not ready for exception
lwip_select: timeout expired
```

cabralfortiss, 2019-01-17:
thanks again for detailed info. We can see that the port 4840 is not getting any packet in your freertos. I looked a little into the tcp header error and it doesn't seem to be a problem (apparently because of wireshark this is happening).

I didn't look too much yet, and I'm trying to follow where does the hostname come from, but I see that the lwip_bind is done on the loopback address (127.0.0.1) although the server is showing 192.168.31.231. I would make sure that this is not the problem by changing something in the open62541 code:
- In the ServerNetworkLayerTCP_start function, there's a call  if(UA_getaddrinfo(NULL, portno, &hints, &res) != 0). Try and change the NULL for "192.168.31.231", recompile and see if it works. 

Also, check if the other working networking task that you say they work, how it shows the debugging LwIP output. See if they also bind to 127.0.0.1 or not. 

Also, see if there's some change in the debug output when you try to connect to the opcuaserver

embbo, 2019-01-17:
From my side i replaced NULL with my IP , but it doesn't work.

FishEatingCat, 2019-01-18:
YES
After changing the NULL of `if(UA_getaddrinfo(NULL, portno, &hints, &res) != 0) `to my IP address, the opc ua server is successfully connected.

FishEatingCat, 2019-01-18:
Now,i want to add mbedtls for encryption, but I have a new problem.
This is one of the compilation conditions for arm mbedtls I am using.
```
#if !defined(MBEDTLS_NO_PLATFORM_ENTROPY)

#if !defined(unix) && !defined(__unix__) && !defined(__unix) && \
    !defined(__APPLE__) && !defined(_WIN32) && !defined(__QNXNTO__)
#error "Platform entropy sources only work on Unix and Windows, see MBEDTLS_NO_PLATFORM_ENTROPY in config.h"
#endif

#if defined(_WIN32) && !defined(EFIX64) && !defined(EFI32)

#if !defined(_WIN32_WINNT)
#define _WIN32_WINNT 0x0400
#endif
#include <windows.h>
#include <wincrypt.h>

int mbedtls_platform_entropy_poll( void *data, unsigned char *output, size_t len,
                           size_t *olen )
{
    HCRYPTPROV provider;
    ((void) data);
    *olen = 0;

    if( CryptAcquireContext( &provider, NULL, NULL,
                              PROV_RSA_FULL, CRYPT_VERIFYCONTEXT ) == FALSE )
    {
        return( MBEDTLS_ERR_ENTROPY_SOURCE_FAILED );
    }

    if( CryptGenRandom( provider, (DWORD) len, output ) == FALSE )
    {
        CryptReleaseContext( provider, 0 );
        return( MBEDTLS_ERR_ENTROPY_SOURCE_FAILED );
    }

    CryptReleaseContext( provider, 0 );
    *olen = len;

    return( 0 );
}
#else /* _WIN32 && !EFIX64 && !EFI32 */
```
However, using mbedtls_`platform_entropy_poll()` in ua_securitypolicy_basicsha.c, what should I do?
```
    /* Add the system entropy source */
    mbedErr = mbedtls_entropy_add_source(&pc->entropyContext,
                                         mbedtls_platform_entropy_poll, NULL, 0,
                                         MBEDTLS_ENTROPY_SOURCE_STRONG);
    UA_MBEDTLS_ERRORHANDLING(UA_STATUSCODE_BADSECURITYCHECKSFAILED);
    if(retval != UA_STATUSCODE_GOOD)
        goto error;

```

cabralfortiss, 2019-01-18:
> 
> 
> YES！
> After changing the NULL of `if(UA_getaddrinfo(NULL, portno, &hints, &res) != 0) `to my IP address, the opc ua server is successfully connected.

Great! that's good to read! This solution is just a workaround, but a proper one should be implemented. I'm currently not working with freeRTOS anymore so I cannot look into it right now. 

> 
> 
> Now,i want to add mbedtls for encryption, but I have a new problem.
> This is one of the compilation conditions for arm mbedtls I am using.
> 
> ```
> #if !defined(MBEDTLS_NO_PLATFORM_ENTROPY)
> 
> #if !defined(unix) && !defined(__unix__) && !defined(__unix) && \
>     !defined(__APPLE__) && !defined(_WIN32) && !defined(__QNXNTO__)
> #error "Platform entropy sources only work on Unix and Windows, see MBEDTLS_NO_PLATFORM_ENTROPY in config.h"
> #endif
> 
> #if defined(_WIN32) && !defined(EFIX64) && !defined(EFI32)
> 
> #if !defined(_WIN32_WINNT)
> #define _WIN32_WINNT 0x0400
> #endif
> #include <windows.h>
> #include <wincrypt.h>
> 
> int mbedtls_platform_entropy_poll( void *data, unsigned char *output, size_t len,
>                            size_t *olen )
> {
>     HCRYPTPROV provider;
>     ((void) data);
>     *olen = 0;
> 
>     if( CryptAcquireContext( &provider, NULL, NULL,
>                               PROV_RSA_FULL, CRYPT_VERIFYCONTEXT ) == FALSE )
>     {
>         return( MBEDTLS_ERR_ENTROPY_SOURCE_FAILED );
>     }
> 
>     if( CryptGenRandom( provider, (DWORD) len, output ) == FALSE )
>     {
>         CryptReleaseContext( provider, 0 );
>         return( MBEDTLS_ERR_ENTROPY_SOURCE_FAILED );
>     }
> 
>     CryptReleaseContext( provider, 0 );
>     *olen = len;
> 
>     return( 0 );
> }
> #else /* _WIN32 && !EFIX64 && !EFI32 */
> ```
> 
> However, this `mbedtls_platform_entropy_poll()` used in ua_securitypolicy_basicsha.c, what should I do?
> 
> ```
>     /* Add the system entropy source */
>     mbedErr = mbedtls_entropy_add_source(&pc->entropyContext,
>                                          mbedtls_platform_entropy_poll, NULL, 0,
>                                          MBEDTLS_ENTROPY_SOURCE_STRONG);
>     UA_MBEDTLS_ERRORHANDLING(UA_STATUSCODE_BADSECURITYCHECKSFAILED);
>     if(retval != UA_STATUSCODE_GOOD)
>         goto error;
> ```

I have no clue about encryption, but you should post this question in another issue, so this ones is kept just as the connection/IP problem, and can be closed soon. 

> 
> 
> From my side i replaced NULL with my IP , but it doesn't work.

It's hard to know with the little information you give. Post some debug information, packets, what's happening, what is the working state, what's the not working state.

Did you manage to compile with the UA_DateTime UA_DateTime_nowMonotonic function?

FishEatingCat, 2019-01-21:
Thank you very much.
I will continue to try to migrate mbedtls and use encryption algorithms to post the issue after getting enough information.

embbo, 2019-01-21:
@FishEatingCat  
Did you manage to compile with the UA_DateTime UA_DateTime_nowMonotonic function?

FishEatingCat, 2019-01-22:
> @FishEatingCat
> Did you manage to compile with the UA_DateTime UA_DateTime_nowMonotonic function?

Of course, I used the RTC time of stm instead of reimplementing UA_DateTime UA_DateTime_nowMonotonic

embbo, 2019-01-25:
@FishEatingCat , Can to discribe little bit how did you use this?
and yo u if you guys have any clue for Non  RTC device.

jpfr, 2019-01-25:
Should these definitions be integrated here?
https://github.com/open62541/open62541/blob/master/arch/freertosLWIP/ua_clock.c

FishEatingCat, 2019-01-26:
It's hard to say, because this involves the hardware driver layer of stm, I don't know if there will be any changes to the whole project after I put it there.

FishEatingCat, 2019-01-28:
Hello!
When I run the STM32 opc ua server, I find that the update data will stop updating after a certain number of times. The ua export prompt error is `BadWaitingForInitialData`, but the client I created with Microsoft Visual Studio reads the data of the stm opc ua server. problem. So, I want to ask what is the problem? How should I solve it?

In addition, after setting the running interval of the opc ua server to 1S, the connection using ua expert will be very stuck. It takes about 18S to complete the connection and display the address space. This is my opc ua server problem or stm hardware problem. Or is it a problem with ua expert? By the way, the client connection created using VS needs about 7S. Of course, these times are my own counts.

1 is the first subscription, then stop not updating, 2 is the second subscription, but my serial port shows the update and then continue
[image]
[image]

embbo, 2019-02-05:
@FishEatingCat hi, If you don't have these functions it still works?
Do we need to change Cmake ?

sammer-h, 2019-02-12:
@FishEatingCat 
I have the same problem with the subscriptions. 
I'm using ESP32 as platform - so I guess not the hardware is the problem  but the the freertos+lwip port?

Subscribing to e.g. the Servers current time node works fine.
But after some seconds the time doesn't get refreshed any more even UAExpert states "Good" statuscode. There is nothing special in the console output (level 100).
The only abnormal is that the heap is decreasing approx. 1-2kb per minute. The task stack size stays const.

FishEatingCat, 2019-02-14:
> @FishEatingCat hi, If you don't have these functions it still works?
> Do we need to change Cmake ?

If you don't have these functions, the compilation will not pass. So, you can try creating an empty function to test.

I have not found a definition of these functions in Cmake, there are implementations of these functions in the ua_clock.c file.

FishEatingCat, 2019-02-14:
> UAExpert

I did not test the subscriptions on the ESP-32.

But my situation on stm32 is the same as you.

After selecting the data you want to view in the address space window of UAExpert, you can see the latest data in the attributes window.

cabralfortiss, 2019-02-14:
I had the same issue with the subscription when I tried. I was using a stm32. In my case if I disconnected and connected again the subscription worked again for some seconds (just to compare with your cases, not saying it's a solution). 

Regarding the time, I think I tried once with the empty function, but from what I remember the connection failed because of some token, or connection parameters that was always zero because of the missing function, which was treated as a connection error. 

I know that freeRTOs has now its own TCP/IP implementation. HAs anyone tried that?

embbo, 2019-02-14:
I belive , this is Freertos problem what i understand, I din't find any solution but it looks same like this article. https://www.freertos.org/FreeRTOS_Support_Forum_Archive/January_2016/freertos_Task_stops_running_without_reason_a14c8defj.html

embbo, 2019-02-14:
@cabralfortiss  .even you can just remove from Data Access View and add again it will works, you doon't need to disconnect

cabralfortiss, 2019-02-17:
Do you use the standard memory allocation functions from the system or use the ones from the freeRTOS memory mangament code?

embbo, 2019-02-18:
@cabralfortiss , I used standard memory allocation functions.

embbo, 2019-02-19:
@cabralfortiss , is it not possible to provide stm freertos lwip .h/c files like raspberry, windows..  .h/c  files are also available? 
like here https://open62541.org/

cabralfortiss, 2019-02-19:
I think it should be possible from the next version. I'm trying to get rid of the #ifdef UA_ARCHITECTURE_FREERTOSLWIP in the amalgamated files so no external define is needed.

One question to all. What's the actual difference  between:
* UA_DateTime_now
* UA_DateTime_localTimeUtcOffset and
* UA_DateTime_nowMonotonic?

and what are the consequences of having them as one function? (UA_DateTime_nowMonotonic I understand as a counter of time, with no references at all)

StalderT, 2019-02-23:
> One question to all. What's the actual difference between:
>    * UA_DateTime_now
>    * UA_DateTime_localTimeUtcOffset and
>    * UA_DateTime_nowMonotonic?

* UA_DateTime_now : date/time in UTC
* UA_DateTime_localTimeUtcOffset : time difference between local time and UTC time. Used to log time in local time.
* UA_DateTime_nowMonotonic : Used to determine the elapsed time or timeout. Independent of a change in the system time.

cabralfortiss, 2019-02-28:
@embbo can you check #2511?

embbo, 2019-03-04:
@cabralfortiss . I have checked , I did changes in my .c/h (din't make again) .
> myserver works fine
>  I checked subscription with really time data from a sensor and checked it for 20 mins and it really works fine.

cabralfortiss, 2019-03-05:
Sorry, I didn't get your current status?
Do you stil have subscription issues?
Did you use the code from the PR and try it?

embbo, 2019-03-05:
No, it works good. I meant to say , I just make changes  in .c/.h, according to your tutorial and it works really fine. I read it for 30 mins .

embbo, 2019-03-05:
@cabralfortiss but i don't find these three functions, 
				UA_VariableAttributes_clear(&attr);
				UA_NodeId_clear(&myIntegerNodeId);
				UA_QualifiedName_clear(&myIntegerName);

cabralfortiss, 2019-03-05:
what do you mean that you don't find them? I think these functions are generated. I have them in my bin folder under src_generated/ua_types_generated_encoding_binary.h and then also in the generated open62541.h

embbo, 2019-03-05:
okay let me genrate again . becoz it gives error , let me check it again thank you.  but server works fine without theses funtions but hold values.

Pro, 2019-03-08:
Please continue any discussion on STM32 and freeRTOS in #2511

closed 2019-03-08 by Pro

embbo, 2019-04-08:
@FishEatingCat  Did you solve the mbedtls problem ?

zxiang1996, 2020-03-27:
I want to implement the ua server on stm32 so that it has subscription and basic address space, but I don't know how much RAM and ROM are needed for my selection, could you please tell me？thank you

chneyi100, 2020-06-11:
I had the some issues about transplanting OPEN62541 , I compiled open62541 source in ubuntu,  The operation is as follows:
    first，create a file to Store compiled files in open62541 source .
    second, cmake -DUA_ARCHITECTURE=freertosLWIP -DUA_ENABLE_AMALGAMATION=ON -        DUA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNTIONS=ON ../  
    third,  make 
 The above command input is completed, get two files open62541.c and open62541.h,I want to ask if my operation is incorrect.
Because I put these two files in my keilv5 project, it will report a lot of errors. First of all, my chip is STMF7. Before transplanting open62541, I have transplanted Freertos and LWIP, and it is running normally.  The error is as follows:
   "inline" is not a type name
   "UA_DateTime"  is undefined

mohamedalielghoula, 2020-07-10:
hello , 
i follow the instruction of building an opc ua server, 
i have freertos n lwip works fine in my stm32f7 board and the integration of open62541.h/c files is good without errors
when i add the opc server task i cant reach the ping again  . why ???

Maku-chan, 2021-08-15:
I created a server on stm32 and client is my ubuntu PC and I am getting badtimeout errors. I have attached the wireshark results.
[image]

PR #2511 (2019-04-04): WIP: Cleanup of freeRTOS and LwIP
I want to use this PR to fix all issues related to freeRTOS and LwIP: 
fixes #1859 
fixes #1864 
fixes #2227 
fixes #2460. 

Users working with freeRTOS that confirmed the PR works:

- [X] @embbo (https://github.com/open62541/open62541/pull/2511#issuecomment-475769413)
- [ ] @nidhal-mars (not active in the PR)
- [ ] @FishEatingCat (not active in the PR)
- [X] @VladBraz125 (https://github.com/open62541/open62541/pull/2511#issuecomment-479844345)
- [X] @rudoq32 (https://github.com/open62541/open62541/pull/2511#issuecomment-474811430)
- [ ] @sammer-h (not active  [... cut]