open62541/open62541 #2229: How to modify the payload size and what's allowable maximal payload size in OPC UA Server/Client model?

Xiongfeng2016, 2018-11-19:
Currently, i am curious about how to modify the payload size of response message to 1MB when OPC UA Server response to OPC UA Client.

Did any people already try this?  Could you provide some help to me? If possible, I will appreciate it! Thank you very much!

jpfr, 2018-11-19:
The size of the messages of course depends on the content.

You can set an upper limit in the configured network layer:
https://github.com/open62541/open62541/blob/master/include/ua_plugin_network.h#L155

Please reopen if an issue remains.

closed 2018-11-19 by jpfr

Xiongfeng2016, 2018-11-23:
> The size of the messages of course depends on the content.
> 
> You can set an upper limit in the configured network layer:
> https://github.com/open62541/open62541/blob/master/include/ua_plugin_network.h#L155
> 
> Please reopen if an issue remains.
Thank you for your reply. I understand your meaning. If i need to send 1MB size content, what content is used to reach 1MZ? If we only send string or int type data, it is really hard to reach 1MZ.