open62541/open62541 #2236: How to parse AutomationML file using XML Nodeset Compiler?

Xiongfeng2016, 2018-11-23:
How to parse AutomationML file using XML Nodeset Compiler?
If someone can provide some guidence, i will appreciate it!
Thank you very much!
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-11-23:
The XML Nodeset Compiler can not be used to parse any other files than OPC UA NodeSet2 XML.

You need to implement your own parser for that.

See also:
https://open62541.org/doc/current/nodeset_compiler.html

closed 2018-11-23 by Pro

Xiongfeng2016, 2018-11-23:
Thank you for your reply!
I am curious that isob.frauhofer has a tool (AML-OPC UA Converter), which can parse AutomationML file to OPC UA file. The parsed AML file has exactly the same format as the OPC UA file parsed by the compiler.
If i want parse AutomationML file to OPC UA, how to implement my own parser. Can you give me some suggestions or references? Thank you very much!!!

Xiongfeng2016, 2018-11-23:
> The XML Nodeset Compiler can not be used to parse any other files than OPC UA NodeSet2 XML.
> 
> You need to implement your own parser for that.
> 
> See also:
> https://open62541.org/doc/current/nodeset_compiler.html
Thank you for your reply!
I am curious that isob.frauhofer has a tool (AML-OPC UA Converter), which can parse AutomationML file to OPC UA file. The parsed AML file has exactly the same format as the OPC UA file parsed by the compiler.
If i want parse AutomationML file to OPC UA, how to implement my own parser. Can you give me some suggestions or references? Thank you very much!!!

jpfr, 2018-11-23:
The author of the AML2UA service (https://aml2ua.iosb.fraunhofer.de) is one room away from mine.
We were also leading the development of the AutomationML companion spec.

The conversion tool is written in C# based on library by the AutomationML association.
The conversion tool itself is not open source. But you can use it if you are a member of the AutomationML association.

You can send a private message if you are are interested in licensing.

RAMI40, 2019-09-17:
Hallo Julius, 

thank you for the information. I am currently working on my thesis. The thesis involves the implementation of a tool as part of process chain that converts the existing automationml models into opc ua xml that can be further used for opc ua server. 

Is there a possibility for university students to access this tool from IOSB you mentioned above or perhaps possibilty to acquire license to this tool especially for students.

I would highly appreciate your time and reply. Thank you in advance. 

Regards.

zediia, 2023-08-01:
Hello Julius,

I am grateful for the information you provide. At present, I have the same problem as the author above. The nodeset_compiler provided by open62541 cannot correctly parse the relevant aml file. I wonder if there is an opportunity to obtain isob.frauhofer's AML-OPC UA Converter tool to integrate aml models into opcua. 
I am looking forward to your reply! Thank you very much!