open62541/open62541 #2249: Run-time adjustable log levels

mustafacamurli, 2018-11-28:
Description
===========
There may be a possibility that certain software systems can have logger which has run-time adjustable log levels. The stack has log-level feature compile time decided. This can be extended to be adjustable in run-time as well.

Background Information
===========================================
Instead of setting "conf->logger" with function pointer we can set it with object pointer and within each log macro we can call the logger function inside object with the state of the logger object like:

conf->logger = &UA_Log_Stdout_obj;

UA_LOG_WARNING(UA_Logger& logger, UA_LogCategory category, const char *msg, ...) {
#if UA_LOGLEVEL <= 400
    va_list args; va_start(args, msg);
    if (logger)
    {
        logger->log_func(logger, UA_LOGLEVEL_WARNING, category, msg, args);
    }
    va_end(args);
#endif

I know that there are so many places to change, but I hope it make sense at the end.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: All
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-11-28:
Can we do this in a custom logger plugin implementation?
The log level is forwarded to the logger. Your own logger implementation can discard logs where you configure the actual log level at runtime.

This is why we have the plugins.

mustafacamurli, 2018-11-29:
In order to forward the log level to logger, first we need to remove compile flags inside each logger.
Then yes, we can implement a custom logger to handle this feature.
As @Pro has commented, this will cost at run-time, but gives us a flexibility to change log level dynamically after the build.

Pro, 2018-11-29:
In our telco we came to the conclusion, that this functionality should for now be implemented by creating your own logging function. Inside that function you can have your own logic to decide if the current log level should be output or not.

closed 2018-11-29 by Pro