open62541/open62541 #2251: How to use server_basic256sha256 to establish a secure channel?

FishEatingCat, 2018-11-29:
Description
===========
When I use the example/encryption/server_basic256sha256.c of open62541 to create a secure channel, the prompts returned are as follows:
The certificate and key I generated is based on https://blog.csdn.net/zxh2075/article/details/79967227

I want to know how to properly establish the safe channel of open62541.

Thanks very much
Background Information / Reproduction Steps
===========================================
open62541\Debug>open62541.exe server-cert.der server-key.pem

[2018-11-29 17:08:25.628 (UTC+0800)] warn/userland      No CA trust-list provided. Any remote certificate will be accepted.
[2018-11-29 17:08:25.631 (UTC+0800)] warn/securitypolicy        mbedTLS returned an error: PK - Invalid key tag or value
[2018-11-29 17:08:25.631 (UTC+0800)] error/securitypolicy       Could not create securityContext
[2018-11-29 17:08:25.634 (UTC+0800)] fatal/userland     Could not create the server config

Checklist
=========
Please provide the following information:

 - [ *] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

FishEatingCat, 2018-12-04:
> Did you make sure that the key you are passing is null terminated? Try adding a null byte to the end of the string.

Hi
As you said, I added a null byte to the end of server.key, but the result is still the same error. 
Thank you for your answer.

Infinity95, 2018-12-04:
Would it be possible for you to provide a fresh key the way you generated yours for me to test?

FishEatingCat, 2018-12-06:
This is the first time I generated the certificate and key.

openssl genrsa -out ca/ca-key.pem 2048
openssl req -new -out ca/ca-req.csr -key ca/ca-key.pem
openssl x509 -req -extfile C:\openssl-1.0.2q\apps\openssl.cnf -extensions v3_req -in ca/ca-req.csr -out ca/ca-cert.pem -signkey ca/ca-key.pem -CAcreateserial -days 3650
openssl x509 -outform der -in ca/ca-cert.pem -out ca/ca-cert.der
openssl pkcs12 -export -clcerts -in ca/ca-cert.pem -inkey ca/ca-key.pem -out ca/ca.p12

openssl genrsa -out server/server-key.pem 2048
openssl req -new -out server/server-req.csr -key server/server-key.pem
openssl x509 -req -extfile C:\openssl-1.0.2q\apps\openssl.cnf -extensions v3_req -in server/server-req.csr -out server/server-cert.pem -signkey ca/ca-key.pem -CAcreateserial -days 3650
openssl x509 -outform der -in server/server-cert.pem -out server/server-cert.der
openssl pkcs12 -export -clcerts -in server/server-cert.pem -inkey server/server-key.pem -out server/server.p12

openssl genrsa -out client/client-key.pem 2048
openssl req -new -out client/client-req.csr -key client/client-key.pem
openssl x509 -req -extfile C:\openssl-1.0.2q\apps\openssl.cnf -extensions v3_req -in client/client-req.csr -out client/client-cert.pem -signkey ca/ca-key.pem -CAcreateserial -days 3650
openssl x509 -outform der -in client/client-cert.pem -out client/client-cert.der
openssl pkcs12 -export -clcerts -in client/client-cert.pem -inkey client/client-key.pem -out client/client.p12

FishEatingCat, 2018-12-06:
The second time I used the creation steps of this website to produce the certificate https://readthedocs.web.cern.ch/display/ICKB/Security
Of course, the results still haven't changed. Even I have run python-related routines successfully. Then I still haven't changed the python certificate and key used in my project. It's really bad!

FishEatingCat, 2018-12-07:
Hello there
I used the script that generates the number of frames and the key of the open62541, and I have completely established a secure channel. Thank you very much for your help. I hope that this part of the guide document for open6254 can be added, which makes it easier to use the open62541 security policy.

closed 2018-12-07 by FishEatingCat