open62541/open62541 #2256: Is UDP support available

divneil, 2018-11-29:
Hi,

I could see `ua_transport_udp.c` existing, but, there isn't UA_ServerNetworkLayer.getJobs member.
So, I am assuming that the patches may be still under review, as the state machine is still not existing to call this API. Can you please point me to patches, I can locally take and try it out.

jpfr, 2018-11-30:
UDP is for OPC UA PubSub only.

We had some experimental support for UDP with the client/server OPC UA protocol.
But that was mostly for some speed tests
https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=7407396

**The standard currently does not specify UDP as a transport possibility for client/server.**

closed 2018-11-30 by jpfr

Hanky27, 2018-11-30:
...but there is a working group in progress to put and use methods (remote procedure calls) on OPC UA Pub/Sub. Other as with Client/Server these Methods will synchronous (blocking) and asynchronous. Of course they will be reliable even they are transported via UDP...

divneil, 2018-12-05:
> ...but there is a working group in progress to put and use methods (remote procedure calls) on OPC UA Pub/Sub. Other as with Client/Server these Methods will synchronous (blocking) and asynchronous. Of course they will be reliable even they are transported via UDP...

Thanks for sharing the information on pub/sub + UDP.  :)
https://opcfoundation.org/news/press-releases/opc-foundation-announces-opc-ua-pubsub-release-important-extension-opc-ua-communication-platform/