open62541/open62541 #2257: PubSub: ReaderGroup and DataSetReader are not yet implemented

Fischer-Thomas, 2018-11-30:
Description
===========
Subscriber implementation is missing. Currently only a manual subscriber as an example is available. The building blocks (ReaderGroup, DataSetReader) of a subscriber according to the specification (Part 14) are not yet implemented. Features belonging to those blocks are not implemented, e.g. filtering of NetworkMessages with respect to the publisherID. 

Also as far as I understood receiving PubSub messages is currently implemented by a periodic timer instead of socket events. This can lead to an additional delay before receiving NetworkMessages by the stack.

Please see tutorial_pubsub_subscribe.c, line 144:
`            UA_Server_addRepeatedCallback(server, (UA_ServerCallback)subscriptionPollingCallback,
                                          connection, 100, &subscriptionCallbackId);`

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch

jpfr, 2018-11-30:
Yes. The subscriber is currently only "manual".

We work on a revamp of the network layer to allow for different types of connections in one select call.
This is required if we want client/server and PubSub to coexist on targets without threading.

We are aware of the issue. We do what we can with the vailable resources.
You can of course let us know if you have the possibility to support the development.

opcua-tsn-team-kalycito, 2019-08-21:
Hi @Fischer-Thomas,
This issue has been addressed in the pull requests #2631, #2944 and #2945, which are all merged to the master.

Regards
opcua-tsn-team-kalycito

closed 2021-04-07 by andreasebner