open62541/open62541 #2258: PubSub: Dynamic Configuration of Publisher, Subscriber and DataSets is not yet implemented

Fischer-Thomas, 2018-11-30:
Description
===========
Up to now Publisher, Subscriber and DataSets cannot be created during runtime (UAServer). According to the PubSub specification (Part 14 Chapter 9) it shall be possible to configure PubSub objects within the server by using methods. 

Please add support for the dynamic configuration.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch

jpfr, 2018-11-30:
We have support for that.
You have to activate the build flags UA_ENABLE_PUBSUB_INFORMATIONMODEL and UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS.

PubSub is still moving fast. Will be stabilized for the 0.4 release family.

@andreasebner I checked with the new version of the Utthunga tools.
Something is borked. I could add neither connections nor datasets.
Can we add documentation for working with the Utthunga tools?

I tried a couple combinations for the URL, etc.
But do not have the time to give this a spin with the debugger...

andreasebner, 2018-11-30:
@jpfr I will check out the new version tomorow and write some documentation. Should we add these specific configuration tool doc into the general open62541-doc? 

About the current state: The configuration at runtime is still in a very early state and does not yet support all PubSub elements. I think in the next 2-3 weeks we can provide a open62541-runtime configuration example/tutorial which creates a fully working PubSub configuration.

teccheck, 2025-04-15:
PubSub information model configuration is now supported, therefore I will close this issue

closed 2025-04-15 by teccheck