open62541/open62541 #2270: client unable to connect to DNS URL.

mukunth1987, 2018-12-04:
Description
===========

Client connection is successful only endpoint is given as ip address and port. It doesn't work with hostname.

Background Information / Reproduction Steps
===========================================

When server is started,  I see the below log  as
_info/network        TCP network layer listening on opc.tcp://2030025883:12626_ 

2030025883 --> my system host name

But when client application attempts to connect to the above server,
connection is successful only when IP address and port information is given as endpoint (opc.tcp://10x.xx.xx.xx:12686)

But connection fails if endpoint is given as opc.tcp://2030025883:12626
_Error log :  warn/network    Trying to connect to opc.tcp://2030025883:12626 timed out_.

Infinity95, 2018-12-04:
Are you able to ping the hostname?

mukunth1987, 2018-12-04:
Hello, that is my localhost address. Ping works

mukunth1987, 2018-12-12:
Hello. any update on the above issue ? 
Please let me know if you need more details on the issue.

jpfr, 2018-12-12:
Is `2030025883` actually the domain name?
Sometimes a computer can have a custom hostname set that is actually unknown on the network.

You can try to open a TCP socket with `telnet 2030025883 12626`.
If that fails then the hostname is not valid.

teccheck, 2025-04-15:
There hasn't been any reply, so it seems to me like this issue is abandoned

closed 2025-04-15 by teccheck