open62541/open62541 #2275: Possible bug in method call

juangburgos, 2018-12-05:
I believe the second `&` might be typo (between `(uintptr_t)` and `object` ):

https://github.com/open62541/open62541/blob/102b7dc69395a9f6e93cd35884b1295f30c6d2d4/src/server/ua_services_call.c#L225

Because it sends a *pointer to a pointer* and it forces the user to *dereference* the `objectContext` in the **method callback** as follows:

```c++
const TYPE * p_ObjContext = (const TYPE *)(*(uintptr_t*)objectContext);
```

Whereas without the `&`, the *dereference* would be much more intuitive:

```c++
const TYPE * p_ObjContext = (const TYPE *)objectContext;
```

It took me a while to figure out why I couldn't dereference the object context correctly.

ccvca, 2018-12-06:
Comparing it to the method-context above, it looks suspicious:
https://github.com/open62541/open62541/blob/102b7dc69395a9f6e93cd35884b1295f30c6d2d4/src/server/ua_services_call.c#L222-L229

I would say this is a bug.

jpfr, 2018-12-06:
Ouch.

Fixed for master and 0.3 in #2278 and #2279.