open62541/open62541 #2286: Custom location of mbedtls

kjeldflarup, 2018-12-10:
When building open62541 cross compiled, I do not have mbedtls as a part of the SDK, but compile it separately.
I get this error:
```
open62541.c:48443:26: fatal error: mbedtls/x509.h: No such file or directory
 #include <mbedtls/x509.h>
```
I obviously need to install mbedtls, but I do not want to put it into the SDK, as it already has openssl, and we do not want to support customers using mbedtls instead of openssl for everything.

Is there a way to specify a custom location of mbedtls when building open62541?

Is it possible to use openssl instead of mbedtls?

 - [ ] open62541 Version (release number or git tag): e9445c401728ff4e7af17f13d8e9e2e4a779b168
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-12-10:
If somebody were to contribute openssl implementations of the SecurityPolicies, we will surely accept them into the code base.
But this has no priority as mbedTLS is generally easier to use on the difficult (embedded) targets.

Okay to close?

kjeldflarup, 2018-12-11:
I solved the problem by adding the include path to the CPATH environment variable.
```
export CPATH="$SDK/$TARGET/sysroot/usr/include:`pwd`/mbedtls-2.13.0-pcm52/include"
```

closed 2018-12-11 by kjeldflarup