open62541/open62541 #2287: Asynchronous connect with late-joining server

alajovic, 2018-12-10:
I am testing the latest development version of open62541 to see whether issue #1912 has been resolved. Indeed, the crash seems to have been fixed, but the asynchronous client connect still doesn't function as expected. The following is a minimum example that exhibits the problem:

CMakeLists.txt

```
cmake_minimum_required(VERSION 3.0)

include(ExternalProject)

ExternalProject_Add(
  external_open62541
  GIT_REPOSITORY "https://github.com/open62541/open62541.git"
  GIT_TAG "4b5dd2f19edd0f8b9737a2e406ece359a37971be"
  INSTALL_COMMAND ""
)

ExternalProject_Get_Property(external_open62541 binary_dir)
set(OPEN62541_LIBRARY open62541)
set(OPEN62541_LIBRARY_PATH ${binary_dir}/bin/${CMAKE_FIND_LIBRARY_PREFIXES}${OPEN62541_LIBRARY}.a)
add_library(${OPEN62541_LIBRARY} UNKNOWN IMPORTED)
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY IMPORTED_LOCATION ${OPEN62541_LIBRARY_PATH})
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY INTERFACE_INCLUDE_DIRECTORIES ${binary_dir})
add_dependencies(${OPEN62541_LIBRARY} external_open62541)

add_executable(async_problem async_problem.c)
set_property(TARGET async_problem PROPERTY C_STANDARD 99)
target_link_libraries(async_problem ${OPEN62541_LIBRARY})
```

async_problem.c

```
#include <stdio.h>
#include <unistd.h>
#include <open62541.h>

static void
onConnect (UA_Client *client, void *userdata, UA_UInt32 requestId, void *status) {
    printf ("Async connect returned with status code %s\n",
            UA_StatusCode_name (*(UA_StatusCode *) status));
}

static int clientProcess() {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    const char* uri = "opc.tcp://localhost:4840";

    for (unsigned int i = 0; ; i++) {
        UA_StatusCode connectStatus = UA_Client_connect_async(client, uri, onConnect, NULL);
        printf("[%u] UA_Client_connect_async -> %s\n", i, UA_StatusCode_name(connectStatus));

        UA_StatusCode iterateStatus = UA_Client_run_iterate(client, 0);
        printf("[%u] UA_Client_run_iterate -> %s\n", i, UA_StatusCode_name(iterateStatus));

        usleep(100000);
    }

    UA_Client_delete(client);
    return 0;
}

static int serverProcess() {
    UA_Boolean running = true;
    UA_ServerConfig* config = UA_ServerConfig_new_default();
    UA_Server* server = UA_Server_new(config);
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return retval;
}

int main() {
    if (fork()) {
        return clientProcess();
    } else {
        // Delay before starting the server process.
        sleep(6);
        return serverProcess();
    }
}
```

Place the two files side by side into the same directory.

```
mkdir build
cd build
cmake ..
make
./async_problem
```

Upon start, the program immediately forks into two processes. The parent process creates a client, calls UA_Client_connect_async() and then starts a polling loop that periodically calls UA_Client_run_iterate(). The child process waits for a few seconds and then starts the server. The intention is to see whether the client will connect once the server becomes available.

If the delay is small – 2 seconds, for example – I see the following (part of the output is omitted because it simply repeats the last two lines over and over):

```
[0] UA_Client_connect_async -> Good
[0] UA_Client_run_iterate -> Good
[1] UA_Client_connect_async -> Good
[1] UA_Client_run_iterate -> BadConnectionClosed
[2] UA_Client_connect_async -> Good
[2] UA_Client_run_iterate -> BadConnectionClosed
[...]
[19] UA_Client_connect_async -> Good
[19] UA_Client_run_iterate -> BadConnectionClosed
[2018-12-10 13:48:58.073 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://cslnb138:4840/
[20] UA_Client_connect_async -> Good
[2018-12-10 13:48:58.082 (UTC+0100)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[20] UA_Client_run_iterate -> Good
[2018-12-10 13:48:58.083 (UTC+0100)] info/channel       Creating a new SecureChannel
[2018-12-10 13:48:58.083 (UTC+0100)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2018-12-10 13:48:58.083 (UTC+0100)] info/channel       Connection 5 | SecureChannel 1 | Opened SecureChannel
[21] UA_Client_connect_async -> Good
[21] UA_Client_run_iterate -> Good
[22] UA_Client_connect_async -> Good
[22] UA_Client_run_iterate -> Good
[23] UA_Client_connect_async -> Good
[23] UA_Client_run_iterate -> Good
[2018-12-10 13:48:58.383 (UTC+0100)] info/session       Connection 5 | SecureChannel 1 | Session ns=1;g=dcb55175-6853-570b-a0ba-8bf855a57a07 | ActivateSession: Session activated
[24] UA_Client_connect_async -> Good
Async connect returned with status code Good
```

So, eventually, the client connects and the async connection callback is invoked, as seen in the last line. However, if the delay is increased – in my environment, 6 seconds is more than enough – the following happens (again, I have omitted the non-interesting parts of the output, as indicated by [...]):

```
[0] UA_Client_connect_async -> Good
[0] UA_Client_run_iterate -> Good
[1] UA_Client_connect_async -> Good
[1] UA_Client_run_iterate -> BadConnectionClosed
[2] UA_Client_connect_async -> Good
[2] UA_Client_run_iterate -> BadConnectionClosed
[...]
[45] UA_Client_connect_async -> Good
[2018-12-10 13:51:34.824 (UTC+0100)] warn/network       Could not create client socket: Too many open files
[2018-12-10 13:51:34.824 (UTC+0100)] error/network      No connection to server.
[45] UA_Client_run_iterate -> BadConnectionClosed
[2018-12-10 13:51:34.925 (UTC+0100)] warn/network       DNS lookup of localhost failed with error Unknown error
[46] UA_Client_connect_async -> BadConnectionClosed
[2018-12-10 13:51:34.925 (UTC+0100)] error/network      No connection to server.
[46] UA_Client_run_iterate -> BadConnectionClosed
[2018-12-10 13:51:35.025 (UTC+0100)] warn/network       DNS lookup of localhost failed with error Unknown error
[47] UA_Client_connect_async -> BadConnectionClosed
[2018-12-10 13:51:35.025 (UTC+0100)] error/network      No connection to server.
[47] UA_Client_run_iterate -> BadConnectionClosed
[...]
[2018-12-10 13:53:50.502 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://cslnb138:4840/
[...]
[2018-12-10 13:53:51.134 (UTC+0100)] warn/network       DNS lookup of localhost failed with error Unknown error
[66] UA_Client_connect_async -> BadConnectionClosed
[2018-12-10 13:53:51.134 (UTC+0100)] error/network      No connection to server.
[66] UA_Client_run_iterate -> BadConnectionClosed
[2018-12-10 13:53:51.235 (UTC+0100)] warn/network       DNS lookup of localhost failed with error Unknown error
[67] UA_Client_connect_async -> BadConnectionClosed
[2018-12-10 13:53:51.235 (UTC+0100)] error/network      No connection to server.
[67] UA_Client_run_iterate -> BadConnectionClosed
[...]
```

The client never connects even though the server becomes available. The message `Could not create client socket: Too many open files` is particularly conspicuous. The fact that something fishy is happening around that point is also confirmed by valgrind that reports `Warning: invalid file descriptor 1031 in syscall socket()` right before the "Could not create..." warning is printed by open62541.

Environment:

* open62541 development version, master as of today (4b5dd2f19edd0f8b9737a2e406ece359a37971be)
* CentOS 7.2

Infinity95, 2018-12-10:
Without having looked at the code this sounds like we fetch a new file descriptor for each connection attempt but never close it, if the attempt fails.
I'll have a look at the code.

Infinity95, 2018-12-12:
My first intuition was correct. There was a file descriptor leak which caused the getaddrinfo function and the function to create new sockets to fail since all devices were busy. I will open a PR with a fix.

commit 3a92275e (2018-12-12): Fix #2287 caused by fd leak
closed 2018-12-12 by jpfr