open62541/open62541 #2290: [CMAKE] Amalgamation install includes non-existend folder

tobijk1, 2018-12-11:
Description
===========
Starting with commit 885bad0dbf33c609d0b56ca981bc3c392f75c0e6 "Install without amalgamation" the amalgamated open62541.h moved from $includedir/open62541/open62541.h -> $includedir/open62541.h, yet the generated open62541Targets.cmake still lists the folder:

`INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include/open62541;${_IMPORT_PREFIX}/include"`

This is followed by an error in depending projects:

> CMake Error in CMakeLists.txt:
>  Imported target "open62541" includes non-existent path
>    "$includedir/open62541"
>
>  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
>  * The path was deleted, renamed, or moved to another location.
>  * An install or uninstall procedure did not complete successfully.
>  * The installation package was faulty and references files it does not
  provide

Background Information / Reproduction Steps
===========================================
Build open62541 with `-DUA_ENABLE_AMALGAMATION=ON`

Conclusion
=========
Preferrably move back to install the open62541.h in $includedir/open62541 to be consistent with the build without ` -DUA_ENABLE_AMALGAMATION=ON` (which installs its files in that folder)
or
remove the non existent path from `INTERFACE_INCLUDE_DIRECTORIES`

Checklist
=========
Please provide the following information:

 - [X ] open62541 Version (release number or git tag): Starting with 885bad0dbf33c609d0b56ca981bc3c392f75c0e6
 - [X ] Other OPC UA SDKs used (client or server): non - not necessary
 - [X ] Operating system: Linux x86_64, aarch64

Pro, 2018-12-11:
I'd go with removing the non existent path from INTERFACE_INCLUDE_DIRECTORIES if no amalgamation.
Can you create a PR?

ccvca, 2018-12-12:
I've just discovered the same issue.
The behaviour was changed by #2262 when the file open62541.h was moved from include/open62541 to include/

https://github.com/open62541/open62541/pull/2262/files#diff-af3b638bc2a3e6c650974192a53c7291R980

I would also recommend to move it back to include/open62541

ccvca, 2018-12-12:
As described here: https://github.com/open62541/open62541/issues/2290#issuecomment-446542415 I would prefer the include/open62541-location.

ccvca, 2018-12-12:
> Is there a specific reason why it should be in a separate subfolder?
You do not scrumble old (>14days) installations, as the old open62541.h-file is no deletet when installing the new one. So there are two open62541.h-Files, one in include and another in include/open62541/

tobijk1, 2018-12-13:
As PR #2292 was merged, i'm closing this! Bugzillalike: Closing as resolved, fixed!

@ccvca: If you really need the old include style back, i would suggest opening another issue. (I'm in favor of it as well, yet  working around it is easily achieved)

closed 2018-12-13 by tobijk1