open62541/open62541 #2295: Memory leak with asynchronous connect

alajovic, 2018-12-13:
When I use the asynchronous connect feature of the OPC-UA client, valgrind reports a memory leak. The amount of memory leaked seems to be proportional to the number of calls to UA_Client_connect_async().

Test outline: the demonstration program below forks into two processes. The first one creates a client and launces a loop in which it alternates between calls to UA_Client_connect_async() and UA_Client_run_iterate(). A little while later, the second process starts a server. Both the client and the server run for a limited amount of time and then terminate.

Expected behavior: when the server starts, the client establishes a connection and calls the async connect callback. The program waits until both the client and the server have performed a given number of event loop iterations, and then terminates. No memory is left allocated upon program exit.

Observed behavior: almost as expected, with the exception that some memory allocations are not reclaimed before the program termination.

CMakeLists.txt

```
cmake_minimum_required(VERSION 3.0)

include(ExternalProject)

ExternalProject_Add(
  external_open62541
  GIT_REPOSITORY "https://github.com/open62541/open62541.git"
  GIT_TAG "d59a9a5119a0953f5211b7bfc1677038cf547e71"
  INSTALL_COMMAND ""
)

ExternalProject_Get_Property(external_open62541 binary_dir)
set(OPEN62541_LIBRARY open62541)
set(OPEN62541_LIBRARY_PATH ${binary_dir}/bin/${CMAKE_FIND_LIBRARY_PREFIXES}${OPEN62541_LIBRARY}.a)
add_library(${OPEN62541_LIBRARY} UNKNOWN IMPORTED)
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY IMPORTED_LOCATION ${OPEN62541_LIBRARY_PATH})
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY INTERFACE_INCLUDE_DIRECTORIES ${binary_dir})
add_dependencies(${OPEN62541_LIBRARY} external_open62541)

add_executable(memory_leak_demo memory_leak_demo.c)
set_property(TARGET memory_leak_demo PROPERTY C_STANDARD 99)
target_link_libraries(memory_leak_demo ${OPEN62541_LIBRARY})
```

memory_leak_demo.c

```
#include <stdio.h>
#include <unistd.h>
#include <sys/wait.h>
#include <open62541.h>

static void
onConnect (UA_Client *client, void *userdata, UA_UInt32 requestId, void *status) {
    printf ("Async connect returned with status code %s\n",
            UA_StatusCode_name (*(UA_StatusCode *) status));
}

static void clientProcess() {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    const char* uri = "opc.tcp://localhost:4840";
    const int iterations = 20;

    for (unsigned int i = 0; i < iterations; i++) {
        UA_Client_connect_async(client, uri, onConnect, NULL);
        UA_Client_run_iterate(client, 0);
        usleep(250000);
    }

    UA_Client_delete(client);
}

static void serverProcess() {
    UA_ServerConfig* config = UA_ServerConfig_new_default();
    UA_Server* server = UA_Server_new(config);
    UA_Server_run_startup(server);

    // Run the server for 100 iterations = 5 seconds.
    for (int i = 0; i < 100; i++) {
        UA_Server_run_iterate(server, true);
    }

    UA_Server_run_shutdown(server);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
}

int main() {
    if (fork()) {
        clientProcess();
        waitpid(-1, NULL, 0);
    } else {
        // Delay before starting the server process.
        sleep(2);
        serverProcess();
    }
    return 0;
}
```

Put the files side to side in a directory, and run

```
mkdir build
cd build
cmake ..
valgrind ./memory_leak_demo
```

In my case, valgrind reports

```
definitely lost: 800 bytes in 10 blocks
```

The number of lost bytes increases if I increase the number of client iterations in line 15, indicating that each call to UA_Client_connect_async() probably creates a new allocation that overwrites the old one.

I have also tried placing the call to UA_Client_connect_async() _before_ the client loop, that is, I tried to move it from line 18 to line 16 so that it is called only once. In such a case, valgrind reports no leaked memory, but the client does not connect when the server becomes available, so the program is no longer behaving as expected.

Environment:
* open62541 development version, master as of today (d59a9a5119a0953f5211b7bfc1677038cf547e71)
* CentOS 7.2

jpfr, 2018-12-13:
Can you run valgrind with --leak-check=full?
Ideally we add a unit test to prevent regressions in the future.

alajovic, 2018-12-14:
Sure. Apologies for not including the output in the initial report. Here it is. PID 27048 is the server process (no leaks there) and PID 27047 is the client process.

```
==27048== 
==27048== HEAP SUMMARY:
==27048==     in use at exit: 0 bytes in 0 blocks
==27048==   total heap usage: 3,055 allocs, 3,055 frees, 1,656,957 bytes allocated
==27048== 
==27048== All heap blocks were freed -- no leaks are possible
==27048== 
==27048== For counts of detected and suppressed errors, rerun with: -v
==27048== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==27047== HEAP SUMMARY:
==27047==     in use at exit: 1,440 bytes in 20 blocks
==27047==   total heap usage: 303 allocs, 283 frees, 887,358 bytes allocated
==27047== 
==27047== 1,440 (800 direct, 640 indirect) bytes in 10 blocks are definitely lost in loss record 2 of 2
==27047==    at 0x4C2DBB6: malloc (vg_replace_malloc.c:299)
==27047==    by 0x44E07B: UA_ClientConnectionTCP_init (ua_network_tcp.c:756)
==27047==    by 0x444A81: UA_Client_connect_async (ua_client_connect_async.c:570)
==27047==    by 0x40390E: clientProcess (in /.../memory_leak_demo)
==27047==    by 0x4039D3: main (in /.../memory_leak_demo)
==27047== 
==27047== LEAK SUMMARY:
==27047==    definitely lost: 800 bytes in 10 blocks
==27047==    indirectly lost: 640 bytes in 10 blocks
==27047==      possibly lost: 0 bytes in 0 blocks
==27047==    still reachable: 0 bytes in 0 blocks
==27047==         suppressed: 0 bytes in 0 blocks
==27047== 
==27047== For counts of detected and suppressed errors, rerun with: -v
==27047== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

jpfr, 2018-12-14:
A fix is proposed in #2297.
But is still untested.

alajovic, 2018-12-21:
The proposed fix does not build.

alajovic, 2018-12-21:
See CI output:
https://travis-ci.org/open62541/open62541/jobs/467959805

closed 2019-01-06 by jpfr

alajovic, 2019-01-16:
This has not been entirely fixed. I still get two blocks of memory marked "definitely lost" with the latest version of open62541. To reproduce this, replace the `ExternalProject_Add` block in the CMakeLists.txt from my initial report with the following:

```
ExternalProject_Add(
  external_open62541
  GIT_REPOSITORY "https://github.com/open62541/open62541.git"
  GIT_TAG "fe01551f83df6f2b466a17e7695d78640ded6289"
  CMAKE_CACHE_DEFAULT_ARGS -DUA_ENABLE_AMALGAMATION:BOOL=ON
  INSTALL_COMMAND ""
)
```

(Note: Even though amalgamated build is no longer the default in recent versions, but I switched it on here, because this way I did not need to change the program source and make extensive modifications to CMakeLists.txt).

Valgrind reports:

```
==11313== 
==11313== HEAP SUMMARY:
==11313==     in use at exit: 0 bytes in 0 blocks
==11313==   total heap usage: 3,025 allocs, 3,025 frees, 1,462,154 bytes allocated
==11313== 
==11313== All heap blocks were freed -- no leaks are possible
==11313== 
==11313== For counts of detected and suppressed errors, rerun with: -v
==11313== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==11312== 
==11312== HEAP SUMMARY:
==11312==     in use at exit: 288 bytes in 4 blocks
==11312==   total heap usage: 202 allocs, 198 frees, 947,775 bytes allocated
==11312== 
==11312== 288 (160 direct, 128 indirect) bytes in 2 blocks are definitely lost in loss record 2 of 2
==11312==    at 0x4C2DBB6: malloc (vg_replace_malloc.c:299)
==11312==    by 0x4707BE: UA_ClientConnectionTCP_init (open62541.c:51766)
==11312==    by 0x45CEF5: UA_Client_connect_async (open62541.c:43843)
==11312==    by 0x403FDE: clientProcess (in /home/alajovic/tmp/async-problem/build/memory_leak_demo)
==11312==    by 0x4040A3: main (in /home/alajovic/tmp/async-problem/build/memory_leak_demo)
==11312== 
==11312== LEAK SUMMARY:
==11312==    definitely lost: 160 bytes in 2 blocks
==11312==    indirectly lost: 128 bytes in 2 blocks
==11312==      possibly lost: 0 bytes in 0 blocks
==11312==    still reachable: 0 bytes in 0 blocks
==11312==         suppressed: 0 bytes in 0 blocks
==11312== 
==11312== For counts of detected and suppressed errors, rerun with: -v
==11312== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

Again, the first part of the report (PID 11313) is for the server process – no leaks there – and the second (PID 11312) is for the client. 160 bytes in 2 blocks appear to be lost. This is better than before, but still not completely clean.

commit e68ff10a (2019-01-06): Client: Cleanup async connection HEL handshake; Fix #2295 memleak

Squashed fix-commit by @StalderT: rename status to retval fix #2297 build