open62541/open62541 #2300: Question about parseing powerlink XML file using open62541 XML Nodeset complier

Xiongfeng2016, 2018-12-15:
I tried to use open62541 XML Nodeset complier to parse Opc.Ua.POWERLINK.NodeSet2.xml, which is included in github/open62541/deps/nodeset/POWERLINK folder. However, i was not able to get right result.

The content of My cmakelist file is as follows:
_**`# Powerlink requires the full ns0 as basis
if(UA_NAMESPACE_ZERO STREQUAL "FULL")

    # Generate types and namespace for DI
    ua_generate_nodeset_and_datatypes(
        NAME "di"
        FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/OpcUaDiModel.csv"
        FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/Opc.Ua.Di.Types.bsd"
        NAMESPACE_IDX 2
        FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml"
        INTERNAL
    )

    # generate powerlink namespace which is using DI
    ua_generate_nodeset_and_datatypes(
        NAME "powerlink"
        # powerlink does not define custom types. Only generate the nodeset
        FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/POWERLINK/Opc.Ua.POWERLINK.NodeSet2.xml"
        # powerlink depends on the di nodeset, which must be generated before
        DEPENDS "di"
        INTERNAL
    )

    add_example(server_nodeset_plcopen server_nodeset_plcopen.c
                ${PROJECT_BINARY_DIR}/src_generated/ua_types_di_generated.c
                ${PROJECT_BINARY_DIR}/src_generated/ua_namespace_di.c
                ${PROJECT_BINARY_DIR}/src_generated/ua_namespace_powerlink.c)
    add_dependencies(server_nodeset_plcopen open62541-generator-ns-powerlink)
   target_include_directories(server_nodeset_plcopen PRIVATE ${PROJECT_SOURCE_DIR}/src) # needs an internal header

endif()`**_

If someone can reply this question,  i will appreciate it. Thank you very much!!!

Pro, 2018-12-19:
Without further information what you mean by "right result", we can not help you...

Please try to extend your issue information and also stick to the template we provide.

Did you already check our documentation?
https://open62541.org/doc/current/nodeset_compiler.html
Read all of it, especially the last chapter.

Xiongfeng2016, 2018-12-19:
> Without further information what you mean by "right result", we can not help you...
>
> Please try to extend your issue information and also stick to the template we provide.
>
> Did you already check our [...]

Thank you for your reply so much. I am so sorry about that i don't discribe my question in detail.  Based on your suggestion, i update the description of my problem more detailed. Please check it in your convenience~~

jpfr, 2019-01-22:
Can you confirm that #2385 fixed the issue?

matkonnerth, 2019-01-26:
Hi, here's an example generating and building the powerlink nodeset based on master. The powerlink nodeset defines some datatypes, I've added the powerlinkTypes.bsd

Pro, 2019-01-30:
This is fixed with #2404

I tested the example from @matkonnerth (which is now also included in the PR) and it works with that one.

Thanks for the issue report and sorry for the long fixing time.

PR #2404 (2019-01-31): NodesetCompiler: Add support for OptionSet
Fixes #2300 

Also adds example for PowerLink Nodeset.
commit 064cf258 (2019-01-31): NodesetCompiler: Add support for OptionSet

Fixes #2300
closed 2019-01-30 by Pro