open62541/open62541 #2306: PubSub: Cannot compile the tutorial_pubsub_subscribe example

theresa95, 2018-12-18:
Description
===========
I have tried to compile the PubSub tutorial examples. This includes the tutorial_pubsub_publish.c and the tutorial_pubsub_subscribe.c files. After I cloned the repository, I built the library as followed:
```
mkdir build 
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON ..
make
```
First I compiled the publish example which worked:
```
gcc -std=c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_publish.c
```
Then, I tried to compile the subscribe example in the same way like this:
```
gcc -std=c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_subscribe.c
```
This generated lots of errors - all saying "No such file or directory(first error was for the #include "ua_architecture_base.h") Then I tried to compile it with the ua_architecture_base.h and get other missing file or directory error and so on...

I also tried to compile the whole include folder of the open62541, but I also get "No such file or directory" errors: 
```
gcc -std=c99 -I /home/theresa/Desktop/open62541/include -o PubSub_Subscribe_Example open62541.c tutorial_pubsub_subscribe.c
```
Background Information / Reproduction Steps
===========================================
I also noticed that of all examples provided by the open62541 master,
tutorial_pubsub_subscribe.c is the only example which does NOT include
the open62541.h folder. Instead it includes a few other header files:

```
#include "ua_pubsub_networkmessage.h"
#include "ua_log_stdout.h"
#include "ua_server.h"
#include "ua_config_default.h"
#include "ua_pubsub.h"
#include "ua_network_pubsub_udp.h"
#ifdef UA_ENABLE_PUBSUB_ETH_UADP
     #include "ua_network_pubsub_ethernet.h"
#endif
#include "src_generated/ua_types_generated.h"
#include <stdio.h>
#includ <signal.h>
```

So how can I compile and run this example and why does the subscriber example not 
include the open62541 header file?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):  I have cloned the github repository today, so it is the current version of open62541 (do not know the version number)
 - [ ] Other OPC UA SDKs used (client or server): no
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: 
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached: 
 - [x] Critical issue: Subscribe example not working.

Pro, 2018-12-18:
Are other examples working for you, or is it only the pub_sub which is not working?

theresa95, 2018-12-18:
It is only the tutorial_pubsub_subscribe example.

Pro, 2018-12-19:
Pubsub is currently in an experimental stage and under heavy development. Its not tested with amalgamation.

You should be able to build the pub sub if you disable amalgamation and build from source

closed 2018-12-19 by Pro

theresa95, 2018-12-19:
What do you mean by building fom source?

theresa95, 2018-12-19:
I also tried to build the open62541 git repo by:
mkdir build
cd build cmake -DCMAKE_BUILD_TYPE=Release - DBUILD_SHARED_LIBS=ON ..
make

But strangely, without the -DUA_ENABLE_AMALGAMATION=ON parameter, on
open62541.c file is created during the building process.

Maybe you could provide me with some instructions on how exactly to build
it.

Best wishes,
Theresa Lichtenberger

Pro, 2018-12-19:
Did you already read through our documentation, especially https://open62541.org/doc/current/building.html ?

If there's something unclear, please tell us and we will try to improve the doc. Also make sure that you are deleting the build directory if you are changing the Amalgamation setting.

FishEatingCat, 2018-12-28:
> Did you already read through our documentation, especially https://open62541.org/doc/current/building.html ?
> 
> If there's something unclear, please tell us and we will try to improve the doc. Also make sure that you are deleting the build directory if you are changing the Amalgamation setting.
Hello, I have not found the Pubsub option here, or UA_ENABLE_SUBSCRIPTIONS contains the Pub?
[https://open62541.org/doc/current/building.html#detailed-sdk-features](url)

 In addition, I downloaded the latest code, compiled with cmake, only saw UA_ENABLE_SUBSCRIPTIONS and UA_ENABLE_SUBSCRIPTIONS_EVENTS, and did not see the relevant options for Pub.
[image]