open62541/open62541 #2311: /include/open62541/ua_architecture.h:145:10: fatal error: ua_architecture_functions.h: No such file or directory

ccvca, 2018-12-19:
Description
===========
When including <ua_types.h> I git the error:
[...]/include/open62541/ua_architecture.h:145:10: fatal error: ua_architecture_functions.h: No such file or directory
The file ua_architecture_function.h is not installed in [...]include/open62541/

@Pro When I get it right, the file should be added to these lines:
https://github.com/open62541/open62541/blob/20eccdcaf15162f48a372186e15f654d81331f59/CMakeLists.txt#L460-L464

Background Information / Reproduction Steps
===========================================
Install fresh on windows 10, include ua_types.h in any file.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 20eccdcaf15162f48a372186e15f654d81331f59
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

Pro, 2018-12-19:
This should be fixed in the current master branch ( i just merged #2302 which contains many improvements for the install).

The corresponding line is already there:
https://github.com/open62541/open62541/blob/82b6499f37dcb64646536218e0eadb3ef497ea08/CMakeLists.txt#L112

Pro, 2018-12-19:
Oh right. Will be fixed with #2316

closed 2018-12-19 by Pro

Pro, 2018-12-20:
@tobijk1 @ccvca with #2318 we are now also testing every commit if `make install` is complete.

Thanks for all your issue reports and support!

tobijk1, 2018-12-20:
Nice, thanks!

ccvca, 2018-12-20:
Thank you for fixing.

commit ea8cc396 (2018-12-19): CMake: Add missing ua_architecture headers

Fixes #2311
commit 4ab4689f (2018-12-19): CMake: Add missing ua_architecture headers

Fixes #2311
PR #2316 (2018-12-20): CMake: Add missing ua_architecture headers
Fixes #2311
commit d89e9e55 (2018-12-20): CMake: Add missing ua_architecture headers

Fixes #2311