open62541/open62541 #2312: Crash when adding Nodes with Multithreading enabled

Fischer-Thomas, 2018-12-19:
Description
===========
Currently when working with the flag for multithreading enabled there is a problem when adding nodes to the server.
The UA_NodeMap_releaseNode asserts that entry->refCount > 0 failed and shortly after the application crashes.
Please see the attachment for the callstack.

Background Information / Reproduction Steps
===========================================
1. Enable UA_ENABLE_MULTITHREADING when building the server
2. Start the server.
3. Add a node to the server with the second thread
[image]

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-12-19:
Note that Multithreading support is far from being in beta stadium. Probably this issue will not be fixed for a long time, but feel free to create a PR yourself.

You will have many similar issues like this one.
The current recommended way for a multithreaded application is that you use your own mutex to make sure you have mutual exclusion on the `UA_Server` pointer.

teccheck, 2025-04-15:
Multithreading is now supported, if enabled

closed 2025-04-15 by teccheck