open62541/open62541 #2315: Setting O0 for ua_namespace0.c not appropriate

tobijk1, 2018-12-19:
Description
===========
Starting with PR 2302 the optimization for ua_namespace0.c gets overruled by a static rule in the cmake build file:
`set_source_files_properties(${PROJECT_BINARY_DIR}/src_generated/ua_namespace0.c PROPERTIES COMPILE_FLAGS "-fno-stack-protector -O0")`

This O0 contradicts settings chosen by the user, e.g CMAKE_BUILD_TYPE = Release with default CMAKE_C_FLAGS_RELEASE = -O3 -DNDEBUG. Both these options can be set by the user, so taking away the "freedom of choice" seems wrong.

PS: If you want quick builds for some reason, just overwrite CMAKE_{C/CXX}_FLAGS_*

tobijk1, 2018-12-19:
As i was really surprised on this, i tested it with the schamas from https://opcfoundation.org/UA/schemas/1.04/, as i didn't find any newer?! And i cannot reproduce this at all, below is a small accumulations of timings and ram usage for -O{1..3} for the file in question as well as -flto, as that sucked with older comilers. All i'm asking for is a choice (maybe even a "advanced" one people can just set if they care)

> cmake -DCMAKE_BUILD_TYPE=Release \
      -DBUILD_SHARED_LIBS=ON \
      -DUA_ENABLE_DISCOVERY=ON \
      -DUA_ENABLE_METHODCALLS=ON \
      -DUA_ENABLE_NODEMANAGEMENT=ON \
      -DUA_ENABLE_SUBSCRIPTIONS=ON \
      -DUA_ENABLE_DISCOVERY=ON \
      -DUA_ENABLE_PUBSUB=OFF \
      -DUA_ENABLE_ENCRYPTION=ON \
      -DUA_ENABLE_MULTITHREADING=ON \
      -DUA_ENABLE_IMMUTABLE_NODES=ON \
      -DUA_BUILD_SELFSIGNED_CERTIFICATE=ON \
      -DCMAKE_INSTALL_PREFIX=/usr \
      -DLIB_INSTALL_DIR=lib64 \
      -DUA_NAMESPACE_ZERO=FULL \
      ..
>
>-O1
>
>gcc 8.2.1
>real    0m50,246s
>user    1m5,172s
>sys     0m2,540s
>
>~800MB
>
>===
>
>clang 6.0.1
>real    2m59,447s
>user    3m18,325s
>sys     0m2,094s
>
>~250MB
>
>---
>
>-O2
>
>gcc 8.2.1
>real    2m47,394s
>user    3m1,433s
>sys     0m2,473s
>
>~650MB
>
>===
>
>clang 6.0.1
>real    6m43,561s
>user    7m2,710s
>sys     0m2,391s
>
>~400MB
>
>---
>
>-O3
>
>gcc 8.2.1
>real    2m51,545s
>user    3m5,597s
>sys     0m2,576s
>
>~700MB
>
>===
>
>clang 6.0.1
>real    7m34,439s
>user    7m53,873s
>sys     0m2,766s
>
>
>~1100MB
>
>---
>
>Release + -flto
>
>gcc 8.2.1
>real    3m1,230s
>user    3m8,934s
>sys     0m3,122s
>
>~800MB
>
>===
>
>clang 6.0.1
>real    14m14,788s
>user    14m31,388s
>sys     0m2,634s
>
>
>~1200MB
>

Pro, 2018-12-20:
Thank you very much for all the tests!

We had the Problem on Travis and Appveyor, that they both timed out after around 1h since the compilation of NS0 didn't finish.

What is the major advantage if you use O2 instead of O0 for the NS0 file?
E.g. what's the resulting file size difference?

If we add an (advanced) option for every special case then the list of options is getting too big. Thus we need to evaluate the importance of the option.
There's always the possibility to create your own internal fix for that.

tobijk1, 2018-12-20:
Yes of course there is always the possibility to fix this on in my private repo, yet as this is an open source project (and a pretty neat one!) i want to make it a bit better (not just for me but for everyone). If my wording is a bit harsh sometimes, sorry for that! Just to reassure you: I don't want to have my needs fulfilled for every little bit, but i believe we (as the community to this project) should argue about points like this and come to a conclusion. You as a main contributor still have the decision to make at the end, thats why i brought the data in the first place. :)

The compiled lib (everything else compiled in RELEASE mode) goes down from around 11MB (-O0) to ~4,5MB (-O2/-O3) depending on the compiler (clang seems to create smaller binaries in general).

PS: Not sure about what takes Travis, Appveyor and your dev machine that long! The test were run on my dev notebook (~2 years old). Maybe there is yet another flag set making things worse?!

PPS: The timings were measuring a whole  make run: `time make -j N`

Pro, 2018-12-20:
We are happy for any comment and improvements! Don't get me wrong :)
I'm definitely in favour of keeping -O2/-O3 for the NS0 file, but if that would result in a significant longer (> 5 Minutes) compile time we will get many issue reports that the compilation is "stuck".

You can try to create a PR which removes this line and then let's see if travis/appveyor is still able to build everything.

Maybe it's due to the older compiler versions?
Travis uses gcc-4.8 and clang-6.0

tobijk1, 2018-12-20:
This could definitely be gcc 4.8, clang 6.0 is slower than recent gccs yet doable (as seen above where llvm 6 is tested as well). gcc 4.8 seems pretty old, which distribution are you targeting with active support for this as main compiler? RHEL 5.x and SLE11?

Pro, 2018-12-20:
We try to support also very old compilers (Travis currently uses Ubuntu Trusty as basis) since focus of the stack is to support various embedded devices. Some embedded devices rely on very old compilers, sometimes even older than 4.8. Some of them even still use Visual Studio 2008 :scream:

tobijk1, 2018-12-20:
Mh allright, now arguably i throw in that embedded devices will not run full ns0 (Freely admitting, i do not have any proof for or against that point). Would be a shame, as on top of this, cross-compilers ought to be exchangeable :/

jpfr, 2018-12-20:
We could add a positive-list in cmake. Clang, GCC and MSVC cover >90% already.
If we detect a recent version of those we can keep the overall optimization level for the project also for ns0.

closed 2019-01-08 by Pro

tobijk1, 2019-01-08:
Thanks for the change!

PR #2320 (2019-01-08): CI: Remove NS0 hardcoded optimization
Remove hardcoded optimization level for NS0 file. Fixes #2315 

This is for now a test to see how fast everything compiles.