open62541/open62541 #2322: [Discovery] mDNS-Response causes server lags

ccvca, 2018-12-21:
Description
===========
UA_Server_run_iterate could take up to 9 Seconds when a mdns-record is received, which could not be resolved to a hostname (e.g. Illegal AutoIp-Adress 169.254.213.26). During this time the server execution is blocked.

**This offers DDOS-possibilities by sending malicous mdns-Packages.**

Background Information / Reproduction Steps
===========================================
I've catched the code path:

Here the mdns-Part is executed inside UA_Server_run_iterate :
https://github.com/open62541/open62541/blob/f523384344987a167b07d6d9cfd4b4ae844c0090/src/server/ua_server.c#L424-L437

When a new record is received the callback 'mdns_record_received' calls setSrv which tries to resolve the adress:
https://github.com/open62541/open62541/blob/f523384344987a167b07d6d9cfd4b4ae844c0090/src/server/ua_server_discovery_mdns.c#L482-L485
This command could take several seconds, when there is no hostname availiable.

I've added some logs before and after:
```
UA_LOG_DEBUG(&server->config.logger, UA_LOGCATEGORY_SERVER, "Before getnameinfo");
        int res = getnameinfo((struct sockaddr*)&sockaddr,
                              sizeof(struct sockaddr_storage),
                              remote_name, sizeof(remote_name),
                              NULL, 0, NI_NAMEREQD);
        UA_LOG_DEBUG(&server->config.logger, UA_LOGCATEGORY_SERVER, "After getnameinfo");
```
Which results in 
```
[2018-12-21 13:56:21.307 (UTC+0100)] debug/server       Before getnameinfo
[2018-12-21 13:56:30.314 (UTC+0100)] debug/server       After getnameinfo
```

Possible Solutions?
===========================================
- Do not resolve the hostname and leave it up to the user to do this
- Use an asynchronous version of getnameinfo which is not portable (maybe getaddrinfo_a (linux), WSAAsyncGetHostByAddr (Windows) )
- Add an option in the config, to disable/enable hostname lookup for those who only need the IP

I would prefer the 1st or 3rd option.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 20eccdcaf15162f48a372186e15f654d81331f59
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Windows 10, Mingw-w64
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue: DDOS

Pro, 2018-12-21:
Yes, that's definitely a valid issue! It's already almost over 3 years when I wrote that code (and to be honest it's not my proudest code, but it works ;) )

The comment says:
https://github.com/open62541/open62541/blob/f523384344987a167b07d6d9cfd4b4ae844c0090/src/server/ua_server_discovery_mdns.c#L459-L467

And it was changed with this PR:
https://github.com/open62541/open62541/pull/1755

I need to do more tests and see if it needs to be there. If yes then we may need some async resolving.

jpfr, 2018-12-24:
Better handling of concurrent network sockets will come with #2271.
Then we can also move all the arch-specific mdns stuff from the core library to /arch or /plugin.

Pro, 2019-01-04:
This will be solved with #2344

commit 1bf487ab (2019-01-04): mDNS: Remove resolving IP address for mDNS announce

Fixes #2322
Fixes #2212
closed 2019-01-04 by Pro