open62541/open62541 #2330: historical_data test randomly crash (SEGFAULT)

StalderT, 2018-12-28:
Description
===========

historical_data test randomly crash (SEGFAULT) see : https://travis-ci.org/open62541/open62541/jobs/473063184#L9687

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): master

Pro, 2019-01-03:
See also https://travis-ci.org/open62541/open62541/jobs/474423872#L8280

Pro, 2019-01-03:
[image]

Found the issue. There's a race condition:
`startPoll` is creating a monitored item, and `stopPoll` is deleting it. Meanwile `serverloop` is executing `UA_Server_run_iterate` which is accessing the deleted timer object for the monitored item.

I'll create a fix for that.

commit 74ee393d (2019-01-03): Tests: Mutex access to server to avoid race condition.

Fixes #2330
commit c90eec35 (2019-01-03): Tests: Mutex access to server to avoid race condition.

Fixes #2330
PR #2342 (2019-01-03): Fix Historical Data Unit Test random failure
Fixes #2330

See https://github.com/open62541/open62541/issues/2330#issuecomment-451135876 for a detailed error description
commit 96e13659 (2019-01-03): Tests: Mutex access to server to avoid race condition.

Fixes #2330
closed 2019-01-03 by Pro