open62541/open62541 #2336: PLC nodeset fails to compile

MPJansen, 2019-01-02:
Description
===========
The DI/PLC nodeset example fails to compile for me with the following errors:

> ua_types_di_generated_encoding_binary.h(16): 
> error C2664: 'size_t UA_calcSizeBinary(void *,const UA_DataType *)': cannot convert argument 1 from 'const UA_DeviceHealthEnumeration *' to 'void *'
> ua_types_di_generated_encoding_binary.h(16): note: Conversion loses qualifiers

and 

> ua_types_di_generated_encoding_binary.h(24): error C2660: 'UA_decodeBinary': function does not take 5 arguments
> ua_namespace_plc.h(25): note: see declaration of 'UA_decodeBinary'

Background Information / Reproduction Steps
===========================================
I followed the steps as per [here](https://open62541.org/doc/current/nodeset_compiler.html)

I suspect the first error has to do with the following line in ua_namespace_:
`size_t
UA_calcSizeBinary(void *p, const UA_DataType *type);`

Which is most likely caused by line 182-183 of backend_open62541.py, if I manually add the const qualifier, this compiler error goes away.

The second error is probably caused by line 204 of generate_datatypes.py, which might be missing an argument? If I add null to all the relevant function calls the compiler error goes away, but I dont know if that solution is correct:

`    return UA_decodeBinary(src, offset, dst, &UA_TYPES_DI[UA_TYPES_DI_DEVICEHEALTHENUMERATION], NULL, NULL);
`

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
Commit: 844d709f402ef5440bf63b5decc34b6baa5d1fd6 
Which is master from 12 days ago.
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
Windows 10 64 bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
N.A.
 - [ ] Wireshark network dump attached
N.A.
 - [ ] Self-contained code example attached
 - [x] Critical issue
No

Please let me know if any information is missing.
Thanks for your awesome work.

KR.

MPJansen, 2019-01-03:
Possibly related to https://github.com/open62541/open62541/issues/1914#issuecomment-417212566, he seems to have encountered a similar compiler error.

Pro, 2019-01-15:
@MPJansen sorry for the late reply!

Could you test with the current master? We merged #2337 which removed the UA_encodeBinary from the nodeset generator.

If that doesn't work, can you post the CMake lines (or python parameters) you used to generate the nodeset?

Pro, 2019-01-30:
@MPJansen I just tested again on current master and I can compile the `open62541/examples/nodeset/server_nodeset_plcopen.c`

Please reopen this issue if you still encounter problems and post your compiler you are using. Also try to clean your build directory.

closed 2019-01-30 by Pro

MPJansen, 2019-01-30:
Hi @Pro,

apologies for not getting back to you on this. I havent been able to return to working on this.
Thanks for taking the time to look at it though!

Kr,