open62541/open62541 #2340: List Profiles, Facets, and Conformance Units in FEATURES.md

Pro, 2019-01-03:
We should have a full list of at least Profiles and Facets in our FEATURES.md table.
This gives a good overview which features are still missing and which are supported by our stack.

See also https://opcfoundation-onlineapplications.org/profilereporting/

basyskom-meerkoetter, 2019-01-09:
@Pro how would you tackle creating this list?

Pro, 2019-01-09:
I'm not that deep in the profiles/facets stuff, but I guess there are two possible ways:

* Manually create the list
   Look at all the possible options and add them manually to the Features.md file. This is some effort since the same facets may be listed in multiple profiles. This could lead to inconsistencies, and on every new release of the spec we have to update the list.
* Automatically create the list
   I'm not sure if there is some .XML file or some machine-readable format which lists all the possible profiles and the corresponding features. If yes, then we should take that file and have an additional config (preferably JSON) file which matches the supported facets/profiles of our stack with that format. The output of that script can be the FEATURES.md file.

Would you have time to do some research what is possible and post your findings here?

mlang-de, 2019-01-31:
Maybe the collection of the tested Conformance Unit in the CTT selection file could be used?
https://github.com/open62541/open62541-ctt/blob/master/open62541.selection.xml
Still missing would be the Facets and profiles.

Pro, 2019-01-31:
Maybe we have some similar files inside the CTT (which is not completely on Github) which contain the full list of conformance units, facets, and profiles. Then we could create a script which transforms this to a .md

mlang-de, 2019-01-31:
This is true. In the file uaprofile.xml are the Profiles and Facets with the required ConformanceUnit are listed.

jpfr, 2019-02-01:
I readded open62541-ctt with just the configuration files.

https://github.com/open62541/open62541-ctt

It needs an update. But we can track the supported facets there.
By having all known-good facets activated in the config.

basyskom-meerkoetter, 2019-02-04:
@jpfr how old is the config you pushed or rather how out if date is it?

open62541.selection.xml is the list of all the conformance units that were "green" at some point.
I am not sure what open62541.ctt.xml contains (apart from being a serialized form of a QAbstractItemModel).

Is the XML information in that repo to create an overview .md? I assume the "uaprofile.xml" mentioned by @mlang-de is also needed. This file is part of the CTT (and therefore not in the repo)?

Pro, 2019-07-17:
I created a first draft of a tool which outputs a markdown file of all the CTT test results for every unit and profile:

https://github.com/open62541/ua-profiles-md/blob/master/PROFILES.md

The input to generate this file was taken from https://github.com/open62541/open62541-ctt/blob/master/open62541_master.results.xml

I'm happy for any input!
Its a first suggestion, and still needs some bugfixing and improvements.

opcua-tsn-team-kalycito, 2019-08-27:
@jpfr @Pro the latest configuration and result for the release v1.0-rc5 have been uploaded to open62541-ctt repo https://github.com/open62541/open62541-ctt/pull/4/files.

Pro, 2019-08-28:
See https://github.com/open62541/ua-profiles-md/pull/5

Note the output is not yet 100% correct, still need to update the CTT files

closed 2021-09-23 by jpfr