open62541/open62541 #2343: Valgrind memory leaks

SebastianGraeber, 2019-01-04:
## Description
I developed a OPCUA server with open62541 and I received a segmentation fault. I run Valgrind on my application and received the following output:

==10637== 
==10637== HEAP SUMMARY:
==10637==     in use at exit: 1,831,491 bytes in 44,127 blocks
==10637==   total heap usage: 388,673 allocs, 344,546 frees, 402,903,452 bytes allocated
==10637== 
==10637== 12 bytes in 1 blocks are indirectly lost in loss record 1 of 27
==10637==    at 0x402B2B8: malloc (vg_replace_malloc.c:299)
==10637==    by 0x80A7A6B: UA_AccessControl_default (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x80905BD: UA_ServerConfig_new_customBuffer (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D609: UA_ServerConfig_new_minimal (open62541.h:27735)
==10637==    by 0x804D609: main (OPCUA-DAS-SERVER.cpp:634)
==10637== 
==10637== 16 bytes in 1 blocks are indirectly lost in loss record 2 of 27
==10637==    at 0x402B2B8: malloc (vg_replace_malloc.c:299)
==10637==    by 0x80A8A21: UA_Nodestore_default_new (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D609: UA_ServerConfig_new_minimal (open62541.h:27735)
==10637==    by 0x804D609: main (OPCUA-DAS-SERVER.cpp:634)
==10637== 
==10637== 16 bytes in 1 blocks are indirectly lost in loss record 3 of 27
==10637==    at 0x402CF2C: calloc (vg_replace_malloc.c:711)
==10637==    by 0x8052331: UA_Array_new (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D611: main (OPCUA-DAS-SERVER.cpp:636)
==10637== 
==10637== 32 bytes in 1 blocks are indirectly lost in loss record 4 of 27
==10637==    at 0x402B2B8: malloc (vg_replace_malloc.c:299)
==10637==    by 0x80A7BA5: UA_AccessControl_default (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x80905BD: UA_ServerConfig_new_customBuffer (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D609: UA_ServerConfig_new_minimal (open62541.h:27735)
==10637==    by 0x804D609: main (OPCUA-DAS-SERVER.cpp:634)
==10637== 
==10637== 36 bytes in 1 blocks are still reachable in loss record 5 of 27
==10637==    at 0x402B205: malloc (vg_replace_malloc.c:298)
==10637==    by 0x402D0B0: realloc (vg_replace_malloc.c:785)
==10637==    by 0x809B7F4: UA_Node_addReference (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637== 
==10637== 40 bytes in 1 blocks are indirectly lost in loss record 6 of 27
==10637==    at 0x402B2B8: malloc (vg_replace_malloc.c:299)
==10637==    by 0x8097DEF: UA_Timer_addRepeatedCallback (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D611: main (OPCUA-DAS-SERVER.cpp:636)
==10637== 
==10637== 40 bytes in 3 blocks are indirectly lost in loss record 7 of 27
==10637==    at 0x402CF2C: calloc (vg_replace_malloc.c:711)
==10637==    by 0x8094CA7: String_decodeBinary (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637== 
==10637== 48 bytes in 1 blocks are indirectly lost in loss record 8 of 27
==10637==    at 0x402B2B8: malloc (vg_replace_malloc.c:299)
==10637==    by 0x80906A0: UA_ServerConfig_new_customBuffer (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D609: UA_ServerConfig_new_minimal (open62541.h:27735)
==10637==    by 0x804D609: main (OPCUA-DAS-SERVER.cpp:634)
==10637== 
==10637== 72 bytes in 1 blocks are indirectly lost in loss record 9 of 27
==10637==    at 0x402CF2C: calloc (vg_replace_malloc.c:711)
==10637==    by 0x8052331: UA_Array_new (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x80905BD: UA_ServerConfig_new_customBuffer (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D609: UA_ServerConfig_new_minimal (open62541.h:27735)
==10637==    by 0x804D609: main (OPCUA-DAS-SERVER.cpp:634)
==10637== 
==10637== 72 bytes in 1 blocks are indirectly lost in loss record 10 of 27
==10637==    at 0x402CF2C: calloc (vg_replace_malloc.c:711)
==10637==    by 0x80526AC: UA_Array_copy (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
==10637==    by 0x804D609: UA_ServerConfig_new_minimal (open62541.h:27735)
==10637==    by 0x804D609: main (OPCUA-DAS-SERVER.cpp:634)
==10637== 
==10637== 108 bytes in 3 blocks are possibly lost in loss record 11 of 27
==10637==    at 0x402B205: malloc (vg_replace_malloc.c:298)
==10637==    by 0x402D0B0: realloc (vg_replace_malloc.c:785)
==10637==    by 0x809B7F4: UA_Node_addReference (in /home/engineer/workspace/git/OPCUA-DAS-SERVER/OPCUA-DAS-SERVER/Release/OPCUA-DAS-SERVER)
[... 73 lines cut ...]
==10637== LEAK SUMMARY:
==10637==    definitely lost: 1,212 bytes in 2 blocks
==10637==    indirectly lost: 1,677,529 bytes in 40,847 blocks
==10637==      possibly lost: 152,714 bytes in 3,277 blocks
==10637==    still reachable: 36 bytes in 1 blocks
==10637==         suppressed: 0 bytes in 0 blocks
==10637== 
==10637== For counts of detected and suppressed errors, rerun with: -v
==10637== ERROR SUMMARY: 5 errors from 5 contexts (suppressed: 0 from 0)

It seems that the memory leaks are from methods that I call from my main method. It's not a critical issue, but it would be fine to sort it out and fix it.

## Background Information / Reproduction Steps
-
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Git-Revision: 0.3-rc2-566-gd59a9a5
 - [x] Other OPC UA SDKs used (client or server): -
 - [x] Operating system: CentOS 7 - Kernel: 3.10.0-957.rt56.910.el7.x86_64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-01-04:
It looks like you are not calling `UA_ServerConfig_delete(config);`

See
https://github.com/open62541/open62541/blob/d715db105747776e327487117c593aaebc83546b/examples/tutorial_server_firststeps.c#L39L44

Please re-open if this doesn't fix it.

closed 2019-01-04 by Pro

SebastianGraeber, 2019-01-04:
Thanks a lot :-)
Issue fixed.