open62541/open62541 #2345: Cannot build on Windows 10 with option UA_ENABLE_ENCRYPTION turned on.

RavilN, 2019-01-04:
## Description
Using CMake GUI, cannot run successfully neither Configure nor Generate steps.
## Background Information / Reproduction Steps
Cloned source code for open62541 and submodules. 
Opened Cmake GUI, and turned on the UA_ENABNLE_ENCRYPTION option. Tried to run configure - it complains that:
Could NOT find MBEDTLS (missing: MBEDTLS_INCLUDE_DIRS MBEDTLS_LIBRARY MBEDX509_LIBRARY MBEDCRYPTO_LIBRARY)

Tried to add those options (they are not listed). When I try to run either Configure or Generate, still gives the same error message, and added options magically disappear from the list of options.

It is also not clear what should be values for options MBEDX509_LIBRARY and MBEDCRYPTO_LIBRARY - in my case MbedTLS produces only one library file, named mbedTls.lib. 
## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): latest version on master branch (by Jan 04, 2019)
 - [ ] Other OPC UA SDKs used (client or server): No
 - [ ] Operating system: Windows 10 Pro
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2019-01-10:
I was able to build open62541 with encryption on windows.
After building mbedtls with cmake, I used 
cmake --build . --target install
 to install it (you may want to alter the path). Which was then found by open62541 when setting the same install directory.

Pro, 2019-01-10:
You can also check out this script to see how we install mbedtls on Appveyor.

https://github.com/open62541/open62541/blob/master/tools/appveyor/install.ps1

And the build scripts:
https://github.com/open62541/open62541/blob/master/tools/appveyor/build.ps1

Please reopen if you still have issues compiling

closed 2019-01-10 by Pro

davy7125, 2019-02-12:
I also failed to build OpcUA with the option UA_ENABLE_ENCRYPTION. When I'm using the following command line

`cmake .. -G "Visual Studio 14 2015" -DUA_ENABLE_ENCRYPTION=ON`

I get the same error

`Could NOT find MBEDTLS (missing: MBEDTLS_INCLUDE_DIRS MBEDTLS_LIBRARY MBEDX509_LIBRARY MBEDCRYPTO_LIBRARY)`

So I downloaded the sources of MBEDTLS. As suggested I run this command (I adapted "install" to target an existing directory)

`cmake . --build --target install`

The error is

`MSBUILD: error MSB1009: Project file does not exist`

I am quite new with CMAKE but there is a need for the version 0.4.0 comprising the encryption and right now this is not easy to build. Is there a simple way to do it?

schroeder-, 2019-02-12:
I suggest using vckpg to install mbedtls. Installing mbedtls:
` vcpkg install mbedtls:x86-windows-static  `
And then add to your cmake 
`-DCMAKE_TOOLCHAIN_FILE="-DCMAKE_TOOLCHAIN_FILE=[vcpkg root]\scripts\buildsystems\vcpkg.cmake"`
and 
`-DVCPKG_TARGET_TRIPLET="x86-windows-static"`

davy7125, 2019-02-12:
Thank you for the suggestion. I have been successful to build the library with this command (in the open62541 directory):

`cmake .. -G "Visual Studio 14 2015" -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_AMALGAMATION=ON -DMBEDTLS_INCLUDE_DIRS="mbedtsl_output\include" -DMBEDTLS_LIBRARY="mbedtsl_output\library\Release" -DMBEDX509_LIBRARY=mbedtsl_output\library\Release" -DMBEDCRYPTO_LIBRARY="mbedtsl_output\library\Release"`

After having built MbedTls, I have manually overridden the variables needed by Open62541.

RavilN, 2019-02-12:
I did build it from command line too, gave up on configuring CMake from its GUI.
So issue on using CMake GUI still remains, current solution is do not use GUI altogether. Which actually is workaround, not solution. 
Probably due to low priority this issue can be closed with resolution "not going to fix".