open62541/open62541 #2363: Continuous writes to var node via UA_Server_writeValue() API while it is subscribed to data change notification fails after some time

bvinodkumar2008, 2019-01-11:
## Description
On continuously calling **UA_Server_writeValue()** API for updating the opcua variable node in **server.c** and subscribing to this data change notification in **client.c**, i see this error in server **open62541_test: ../open62541.c:48832: UA_NodeMap_releaseNode: Assertion `entry->refCount > 0' failed.**

## Background Information / Reproduction Steps

1. Run server
```sh
gcc -std=c99 open62541.c server.c -o server
./server
```
2. Run client
```sh
gcc -std=c99 open62541.c client.c -o client
./client
```

**server.c**
```
#include <signal.h>
#include "open62541.h"
#include <pthread.h>

static UA_Server *server = NULL;
static UA_Boolean serverRunning = true;
UA_ServerConfig *serverConfig = NULL;
static char dataToPublish[100];
static char errorMsg[100];

void signalHandler(int sig) {
    serverRunning = false;
}

/* This function provides data to the subscriber */
static UA_StatusCode
readPublishedData(UA_Server *server,
                const UA_NodeId *sessionId,
                void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *data) {
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "In readPublishedData() function");
    data->hasValue = true;
    UA_String str = UA_STRING(dataToPublish);
    UA_Variant_setScalarCopy(&data->value, &str, &UA_TYPES[UA_TYPES_STRING]);
	return UA_STATUSCODE_GOOD;
}

static UA_StatusCode
writePublishedData(UA_Server *server,
                 const UA_NodeId *sessionId, void *sessionContext,
                 const UA_NodeId *nodeId, void *nodeContext,
                 const UA_NumericRange *range, const UA_DataValue *data) {

    return UA_STATUSCODE_GOOD;
}

static UA_UInt16
addTopicDataSourceVariable(UA_Server *server, char *namespace, char *topic) {

    UA_UInt16 namespaceIndex = UA_Server_addNamespace(server, namespace);
    if (namespaceIndex == 0) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "UA_Server_addNamespace has failed");
        return 100;
    }
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%u:namespaceIndex created for namespace: %s\n", namespaceIndex, namespace);

    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.description = UA_LOCALIZEDTEXT("en-US", topic);
    attr.displayName = UA_LOCALIZEDTEXT("en-US", topic);
[... 236 lines cut ...]
            /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
            UA_sleep_ms(1000);
            continue;
        }

        UA_Client_run_iterate(client, 1000);
    };

    /* Clean up */
    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}
```

## Checklist

Please provide the following information:

 - [bbafc24568049eb4b28492f20726f9b36e69b296] open62541 Version (release number or git tag):
 - [None] Other OPC UA SDKs used (client or server): 
 - [Ubuntu 16.04] Operating system:
 - [300] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [No] Wireshark network dump attached
 - [Yes] Self-contained code example attached
 - [Yes] Critical issue - continuous writes to variable node

Pro, 2019-01-11:
That's probably because you are using the `server` in multiple threads, i.e. in the `startServer` and `serverPublish` method.

Multithreading is currently not supported. Therefore you need to make sure that you only access the server sequentially. E.g. with mutex or similar.

Please try that and check if the error is stil there.

bvinodkumar2008, 2019-01-11:
Thanks for quick response. I did try acquiring mutex lock before running **UA_Server_writeValue()** API but even with that I was still get this issue. By the way, I had not enabled multithreading feature on the compiled open62451, should I enable and try?

Pro, 2019-01-11:
You need to use a mutext not only for the UA_Server_writeValue, but everywhere, where you are using the `server` variable. Especially you need to change

```
static void*
startServer(void *ptr) {
    /* run server */
    UA_StatusCode retval = UA_Server_run(server, &serverRunning);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "\nServer failed to start, error: %s", UA_StatusCode_name(retval));
    }
    return NULL;
}
```

to smth like:

```
static void*
startServer(void *ptr) {
    /* run server */
   UA_StatusCode retval = UA_Server_run_startup(server);
  // check for GOOD
  while (serverRunning) {
   // lock mutex
   UA_Server_run_iterate(server, 100);
   // unlock mutex
}
UA_Server_run_shutdown(server);
    return NULL;
}
```

Otherwise you still do not have all the server instances mutexed.

The multithreading feature should not be enabled, this is in very alpha state.

bvinodkumar2008, 2019-01-12:
I used your logic in `startServer` but saw that the call was blocked there with just 1 publish, so i've commented out for now. I'm sharing the files with you to see if i'm doing anything wrong in there that could be causing this.

**server.c**:
```
#include <signal.h>
#include "open62541.h"
#include <pthread.h>

static UA_Server *server = NULL;
static UA_Boolean serverRunning = true;
UA_ServerConfig *serverConfig = NULL;
static char dataToPublish[100];
static char errorMsg[100];
pthread_mutex_t serverLock;

void signalHandler(int sig) {
    serverRunning = false;
}

/* This function provides data to the subscriber */
static UA_StatusCode
readPublishedData(UA_Server *server,
                const UA_NodeId *sessionId,
                void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *data) {
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "In readPublishedData() function");
    data->hasValue = true;
    UA_String str = UA_STRING(dataToPublish);
    UA_Variant_setScalarCopy(&data->value, &str, &UA_TYPES[UA_TYPES_STRING]);
	return UA_STATUSCODE_GOOD;
}

static UA_StatusCode
writePublishedData(UA_Server *server,
                 const UA_NodeId *sessionId, void *sessionContext,
                 const UA_NodeId *nodeId, void *nodeContext,
                 const UA_NumericRange *range, const UA_DataValue *data) {

    return UA_STATUSCODE_GOOD;
}

static UA_UInt16
addTopicDataSourceVariable(UA_Server *server, char *namespace, char *topic) {

    UA_UInt16 namespaceIndex = UA_Server_addNamespace(server, namespace);
    if (namespaceIndex == 0) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "UA_Server_addNamespace has failed");
        return 100;
    }
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%u:namespaceIndex created for namespace: %s\n", namespaceIndex, namespace);

    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.description = UA_LOCALIZEDTEXT("en-US", topic);
    attr.displayName = UA_LOCALIZEDTEXT("en-US", topic);
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Add the variable node to the information model */
    UA_NodeId currentNodeId = UA_NODEID_STRING(namespaceIndex, topic);
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(namespaceIndex, topic);
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);

    UA_DataSource topicDataSource;
    topicDataSource.read = readPublishedData;
    topicDataSource.write = writePublishedData;
    UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,
[... 226 lines cut ...]
    UA_Client *client = UA_Client_new(config);

    /* Endless loop runAsync */
    while(running) {
        /* if already connected, this will return GOOD and do nothing */
        /* if the connection is closed/errored, the connection will be reset and then reconnected */
        /* Alternatively you can also use UA_Client_getState to get the current state */
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                         "Not connected. Retrying to connect in 1 second");
            /* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
            /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
            UA_sleep_ms(1000);
            continue;
        }

        UA_Client_run_iterate(client, 1000);
    };

    /* Clean up */
    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}
```

bvinodkumar2008, 2019-01-16:
Hi @Pro ,
The above issue of call blocking was occuring as i wasn't releasing the mutex on success of writeValue API in serverPublish() method. Once i fixed it, the server snippet you shared is working fine but even with this change i'm running into the refCount > 0 assertion issue.  The issue could be easily reproduced with the shared code below. 

Let me know if you want me to try anything else.

**`server.c**:
```sh
#include <signal.h>
#include "open62541.h"
#include <pthread.h>

static UA_Server *server = NULL;
static UA_Boolean serverRunning = true;
UA_ServerConfig *serverConfig = NULL;
static char dataToPublish[100];
static char errorMsg[100];
pthread_mutex_t serverLock;

void signalHandler(int sig) {
    serverRunning = false;
}

/* This function provides data to the subscriber */
static UA_StatusCode
readPublishedData(UA_Server *server,
                const UA_NodeId *sessionId,
                void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *data) {
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "In readPublishedData() function");
    data->hasValue = true;
    UA_String str = UA_STRING(dataToPublish);
    UA_Variant_setScalarCopy(&data->value, &str, &UA_TYPES[UA_TYPES_STRING]);
	return UA_STATUSCODE_GOOD;
}

static UA_StatusCode
writePublishedData(UA_Server *server,
                 const UA_NodeId *sessionId, void *sessionContext,
                 const UA_NodeId *nodeId, void *nodeContext,
                 const UA_NumericRange *range, const UA_DataValue *data) {

    return UA_STATUSCODE_GOOD;
}

static UA_UInt16
addTopicDataSourceVariable(UA_Server *server, char *namespace, char *topic) {

    UA_UInt16 namespaceIndex = UA_Server_addNamespace(server, namespace);
    if (namespaceIndex == 0) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "UA_Server_addNamespace has failed");
        return 100;
    }
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%u:namespaceIndex created for namespace: %s\n", namespaceIndex, namespace);

    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.description = UA_LOCALIZEDTEXT("en-US", topic);
    attr.displayName = UA_LOCALIZEDTEXT("en-US", topic);
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Add the variable node to the information model */
    UA_NodeId currentNodeId = UA_NODEID_STRING(namespaceIndex, topic);
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(namespaceIndex, topic);
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);

    UA_DataSource topicDataSource;
[... 235 lines cut ...]
    UA_Client *client = UA_Client_new(config);

    /* Endless loop runAsync */
    while(running) {
        /* if already connected, this will return GOOD and do nothing */
        /* if the connection is closed/errored, the connection will be reset and then reconnected */
        /* Alternatively you can also use UA_Client_getState to get the current state */
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                         "Not connected. Retrying to connect in 1 second");
            /* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
            /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
            UA_sleep_ms(1000);
            continue;
        }

        UA_Client_run_iterate(client, 1000);
    };

    /* Clean up */
    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}
```

Pro, 2019-01-16:
I could reproduce the issue locally. The server asserts on this line, but only if the client is reading:

https://github.com/open62541/open62541/blob/fe01551f83df6f2b466a17e7695d78640ded6289/plugins/ua_nodestore_default.c#L218

I can not wrap my head around the cause, how this may happen. As far as I can see, your calls are correctly mutexed.

Maybe @jpfr can have a look at it

bvinodkumar2008, 2019-01-18:
Hi @Pro, @jpfr ,

For now, i just added an if check instead of doing the assert at  https://github.com/open62541/open62541/blob/fe01551f83df6f2b466a17e7695d78640ded6289/plugins/ua_nodestore_default.c#L218 like below to get the server and client working. Do you think there would be any side affects of doing so?

Workaround:
```sh
 if (entry->refCount > 0) {
   --entry->refCount;
   cleanupEntry(entry);
 }
```

jpfr, 2019-01-21:
FYI, we have continued this discussion in private communication.
This issue will be updated/resolved when ready.

bvinodkumar2008, 2019-06-04:
Hi @jpfr,
The below workaround that you shared to my intel email id was working without any refCount issues.
```
The locking with mutexes looks correct. So the above stack trace should be impossible.
Maybe there was a memory corruption.

I made two changes to the code:
1.	For setting the new value, the char-array is wrapped in a UA_String structure.
(This could have led to a memory-corruption as UA_String was expected on the other end…)
2.	The mutex is now heap-allocated. To make memory corruptions on the mutex less likely.
This has been running for 45minutes without a problem now.
And the server comes out squaky-clean even with Valgrind instrumentation.

The server code, with major changes highlighted, is pasted here:
http://paste.debian.net/1061757/

We have some ideas to make open62541 thread-safe internally. So that this wrapping with mutexes is no longer required.
But this will require some effort and extensive testing before going into production.
```
But when i tested this with open62541 version (https://github.com/open62541/open62541/tree/cca3ead344186f9c0aac0e177263c48c52e49c9c) and with one additional change in the API i.e., using `UA_Client_MonitoredItems_createDataChanges()` 
instead of `UA_Client_MonitoredItems_createDataChange()`, I'm again running into the refCount issue. 
I'm attaching below the sample codes for server and client below:
# server.c
```
#include <signal.h>
#include "open62541.h"
#include <pthread.h>

static UA_Server *server = NULL;
static UA_Boolean serverRunning = true;
static UA_ServerConfig *serverConfig = NULL;
static char dataToPublish[1000];
static pthread_mutex_t *serverLock = NULL;
static char errorMsg[100];

void signalHandler(int sig) {
    serverRunning = false;
}

char* printTime() {

    int millisec;
    struct tm* tm_info;
    struct timeval tv;

    gettimeofday(&tv, NULL);

    millisec = tv.tv_usec / 1000.0;
    if (millisec >= 1000) {
        millisec -= 1000;
        tv.tv_sec++;
    }

    char buffer[26];
    static char timeStr[100];
    tm_info = localtime(&tv.tv_sec);
    strftime(buffer, 26, "%Y:%m:%d %H:%M:%S", tm_info);
    sprintf(timeStr, "%s.%03d", buffer, millisec);
    return timeStr;
}

/* This function provides data to the subscriber */
static UA_StatusCode
readPublishedData(UA_Server *server,
                const UA_NodeId *sessionId,
                void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *data) {
    UA_LOG_DEBUG(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
[... 636 lines cut ...]
    }
}
```

# Steps to run server and client:
## Server
```
$ gcc -std=c99 server.c open62541.c -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o server
$ ./server /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der
```
## Client
```
$ gcc -std=c99 client.c open62541.c -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o client
$ ./client /etc/ssl/opcua/opcua_client_certificate.der /etc/ssl/opcua/opcua_client_key.der /etc/ssl/ca/ca_certificate.der
```
You can use certs generated by the cert tool (https://github.com/open62541/open62541/tree/cca3ead344186f9c0aac0e177263c48c52e49c9c/tools/certs) for supplying to server and client programs. What i've noticed is, after some 34k+ publishes (in around few mins) of server and client running, server is failing with below refCount issue
```
server: open62541.c:47869: NodeMap_releaseNode: Assertion `entry->refCount > 0' failed.
Aborted (core dumped)
```

Let me know if you need any additional details.

Regards,
Vinod

cochicde, 2020-03-30:
Hi,

in your last example, addTopicDataSourceVariable is inside the loop. Is that intended? Also, this function access the server, but there's no mutex around it

teccheck, 2025-04-15:
Normal report seems to be fixed and much has changed in the code. Also there are no issues in the latest CTT, so I'll close this