open62541/open62541 #2364: Problems connecting to Softing dataFEED OPC UA Demo Server

jkahlert, 2019-01-13:
## Description
Hi,

I'm an OPC UA and Open62541 beginner. When trying to connect to the Softing dataFEED OPC UA Demo Server anonymous by the following code

`
UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:51510/UA/DemoServer");`

I get a return value of UA_STATUSCODE_BADINTERNALERROR. It's no problem to connect the server e. g. by the dataFEED OPC UA Demo Client or the OPC UA Demo Client from Unified Automation.

The corresponding code

`UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:48010");
`

works fine when connecting to the OPC UA Demo Server from Unified Automation or the simple sample server from the Open62541 package.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Current master
 - [ ] Other OPC UA SDKs used (client or server): see above
 - [ ] Operating system: Windows 10

jpfr, 2019-01-13:
The Softing demo server expects the client to encrypt the UserToken (authentication).
This is not yet supported.

This is the log output
```
jpfr@virtualbox:~/software/open62541/build$ ./bin/examples/tutorial_client_firststeps
[2019-01-13 19:54:30.866 (UTC+0100)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-01-13 19:54:30.867 (UTC+0100)] debug/network      Sent HEL message
[2019-01-13 19:54:30.869 (UTC+0100)] debug/network      Received ACK message
[2019-01-13 19:54:30.870 (UTC+0100)] debug/channel      Requesting to open a SecureChannel
[2019-01-13 19:54:30.871 (UTC+0100)] debug/channel      OPN message sent
[2019-01-13 19:54:30.872 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 0 | Decrypting chunk
[2019-01-13 19:54:30.872 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 0 | Chunk size before and after decryption: 135, 135
[2019-01-13 19:54:30.873 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 0 | Verifying chunk signature
[2019-01-13 19:54:30.873 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 0 | Decrypted and verified chunk with request id 1 and sequence number 1
[2019-01-13 19:54:30.873 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 0 | Sequence Number processed: 1
[2019-01-13 19:54:30.874 (UTC+0100)] debug/client       Decode a message of type OpenSecureChannelResponse
[2019-01-13 19:54:30.874 (UTC+0100)] debug/channel      Opened SecureChannel acknowledged by the server
[2019-01-13 19:54:30.874 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 10 | Generating new local keys
[2019-01-13 19:54:30.874 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 10 | Generating new remote keys
[2019-01-13 19:54:30.874 (UTC+0100)] debug/client       Sending a request of type 426
[2019-01-13 19:54:30.881 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 10 | Decrypting chunk
[2019-01-13 19:54:30.881 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 10 | Chunk size before and after decryption: 13368, 13368
[2019-01-13 19:54:30.882 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 10 | Decrypted and verified chunk with request id 2 and sequence number 2
[2019-01-13 19:54:30.882 (UTC+0100)] trace/channel      Connection 3 | SecureChannel 10 | Sequence Number processed: 2
[2019-01-13 19:54:30.882 (UTC+0100)] debug/client       Decode a message of type GetEndpointsResponse
[2019-01-13 19:54:30.882 (UTC+0100)] error/client       No suitable UserTokenPolicy found for the possible endpoints
```

And these are the received UserTokenPolicies
```
(gdb) print endpoint->userIdentityTokens[0]
$6 = {policyId = {length = 1, data = 0x5555556109b0 "0"}, tokenType = UA_USERTOKENTYPE_ANONYMOUS,
  issuedTokenType = {length = 0, data = 0x0}, issuerEndpointUrl = {length = 0, data = 0x0},
  securityPolicyUri = {length = 51,
    data = 0x5555556109d0 "http://opcfoundation.org/UA/SecurityPolicy#Basic256"}}
(gdb) print endpoint->userIdentityTokens[1]
$7 = {policyId = {length = 1, data = 0x555555610a10 "1"}, tokenType = UA_USERTOKENTYPE_USERNAME,
  issuedTokenType = {length = 0, data = 0x0}, issuerEndpointUrl = {length = 0, data = 0x0},
  securityPolicyUri = {length = 51,
    data = 0x555555610a30 "http://opcfoundation.org/UA/SecurityPolicy#Basic256"}}
(gdb) print endpoint->userIdentityTokens[2]
$8 = {policyId = {length = 1, data = 0x555555610a70 "2"}, tokenType = UA_USERTOKENTYPE_CERTIFICATE,
  issuedTokenType = {length = 0, data = 0x0}, issuerEndpointUrl = {length = 0, data = 0x0},
  securityPolicyUri = {length = 51,
    data = 0x5555556114d0 "http://opcfoundation.org/UA/SecurityPolicy#Basic256"}}
```

We have encryption support. But only if the entire message is encrypted.
Not just the UserToken.

Curiously enough, the server expects the "anonymous" UserToken to be encrypted...

Support for this will land in the coming weeks.
The PR #2247 will add necessary infrastructure.

jpfr, 2019-02-10:
Fixed on master

closed 2019-02-10 by jpfr

jkahlert, 2019-02-11:
Do I have to make any changes to my source code to use the new feature? Or
do I only have zo exchange the open62541.c/.h files?

 

Best regards

 

---------------------------------------------

Dr.-Ing. Jörg Kahlert

Ingenieurbüro Dr. Kahlert

Ludwig-Erhard-Str. 45

D-59065 Hamm

Tel. 02381/926996

Fax 02381/926997

www.kahlert.com

  _____  

OPC UA Demo Server (#2364)

 

it on GitHub, or mute
CveeENsks5vMFw9gaJpZM4Z9NvM>  the thread.
<https://github.com/notifications/beacon/AsfUPDONrG5gjiLpicXpVtFar841PpgRks5
vMFw9gaJpZM4Z9NvM.gif>

jkahlert, 2019-02-11:
Do I have to make any changes to my source code to use the new feature? Or do I only have zo exchange the open62541.c/.h files?