open62541/open62541 #2368: OPC UA DI NodeSet cannot be loaded (open62541 v0.3.0)

talsen-team, 2019-01-14:
## Description

Following the instructions in the [open62541 documentation v0.3.0](https://open62541.org/doc/0.3/nodeset_compiler.html#combination-of-multiple-nodesets) the compiled nodeset fails to load.
We created [a small demo project](https://github.com/talsen-team/issue-open62541-opc-ua-di-nodeset-cannot-be-loaded) to reproduce the error.

## Background Information / Reproduction Steps

Just follow the 3 step instruction (which just automate the described steps from the open61542 documentation via bash scripts) to reproduce the error when calling the generated server-app executable.
(more detailed information about what steps are done in detail are found in the [build-and-run.sh](https://github.com/talsen-team/issue-open62541-opc-ua-di-nodeset-cannot-be-loaded/blob/master/build-and-run.sh) file)

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): tag v0.3.0
 - [ ] Other OPC UA SDKs used (client or server): nope
 - [ ] Operating system: Ubuntu 18.04 Desktop LTS (minimal setup)
 - [ ] Dependencies (via apt-get): gcc, meson, python3
 - [ ] ~~Logs (with `UA_LOGLEVEL` set as low as necessary) attached~~  not relevant
 - [ ] ~~Wireshark network dump attached~~ not relevant
 - [ ] Self-contained code example linked via [GitHub](https://github.com/talsen-team/issue-open62541-opc-ua-di-nodeset-cannot-be-loaded)
 - [ ] Maybe critical issue ( at least for us :) )

ccvca, 2019-01-15:
#2336

talsen-team, 2019-01-15:
Sorry for the accidental duplicate :S

Pro, 2019-01-15:
Thank you for your really nice issue report! It would be nice to get such reproducability for every issue :)

For further reference, this is the output of the server:

```
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[2019-01-15 19:07:17.312 (UTC+0100)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[2019-01-15 19:07:17.313 (UTC+0100)] error/server	Adding the DI namespace failed. Please check previous error output.
```

I did not work with meson yet, thus it's hard to tell for me which CMake Options you used. Is it the default options?

There are some parent nodes which are missing (note that on master we already have better output, including the node id).

The default CMake options are set to use a reduced NS0 nodeset which does not contain all the nodes, DI requires.

Can you test to set the following CMake option:

```cmake -DUA_ENABLE_FULL_NS0=ON ..```

See
https://github.com/open62541/open62541/blob/6249bb29de68e22974f2adf232256b63471914ae/CMakeLists.txt#L104

This should generate the full NS0

talsen-team, 2019-01-15:
Thanks for the tip `cmake -DUA_ENABLE_FULL_NS0=ON ..`, it did the thing. I remember stumbling over it in the documentation (now as you mentioned it), but we (our dev team) are not quite into cmake (mostly coming from C# .NET Framework / Core which is quite different to the C/C++ world).

The linked [repo](https://github.com/talsen-team/issue-open62541-opc-ua-di-nodeset-cannot-be-loaded/tree/master) has also been updated, it builds now open62541 and some demo main.c file which loads the DI model successfully.

In our first try we used the [GitHub release assets](https://github.com/open62541/open62541/releases) of tag v0.3.0 (the [open62541.c](https://github.com/open62541/open62541/releases/download/v0.3.0/open62541.c) and [open62541.h](https://github.com/open62541/open62541/releases/download/v0.3.0/open62541.h) files) and ran into the initially presented problem.

PS: we only used meson to build the already existing open62541.c from the GitHub release assets file with the demo main.c file

PPS: for newcomers meson seems to be a bit easier to get into than cmake (at least simple tasks seem more intuitive)

Thanks again and sorry for bothering you ;-)

closed 2019-01-15 by talsen-team