open62541/open62541 #2369: Nodeset compiler for information model that contains the own structure data types

MrPoudel, 2019-01-15:
## Description
I created an information model in UaModeler which contains a variable type node called myVector. The data type of the myVector is derived from the Structure type( which in turn derived from the BaseDataType).

[image]

After that, I exported the information model as an XML file and generated the example_nodeset.c and example_nodeset.h successfully. However, during compilation, I got the following errors [related to the encoding]:

[image]

The content of the "server_nodeset.xml" is as follows:

`<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/server_test/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/server_test/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="String">i=12</Alias>
        <Alias Alias="ByteString">i=15</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasEncoding">i=38</Alias>
        <Alias Alias="HasDescription">i=39</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="Vector">ns=1;i=3002</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ua:ModelInfo Tool="UaModeler" Hash="espQMHElEZmp5olrXxxdpA==" Version="1.5.2"/>
        </Extension>
    </Extensions>
    <UADataType NodeId="ns=1;i=3002" BrowseName="1:Vector">
        <DisplayName>Vector</DisplayName>
        <References>
            <Reference ReferenceType="HasEncoding">ns=1;i=5001</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=5002</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
        </References>
        <Definition Name="1:Vector">
            <Field DataType="Double" Name="X"/>
            <Field DataType="Double" Name="Y"/>
            <Field DataType="Double" Name="Z"/>
        </Definition>
    </UADataType>
    <UAVariable DataType="String" ParentNodeId="ns=1;i=6001" NodeId="ns=1;i=6005" BrowseName="1:Vector">
        <DisplayName>Vector</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=69</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=6001</Reference>
            <Reference ReferenceType="HasDescription" IsForward="false">ns=1;i=5001</Reference>
        </References>
        <Value>
            <uax:String>Vector</uax:String>
        </Value>
    </UAVariable>
    <UAVariable DataType="String" ParentNodeId="ns=1;i=6003" NodeId="ns=1;i=6006" BrowseName="1:Vector">
        <DisplayName>Vector</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=69</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=6003</Reference>
            <Reference ReferenceType="HasDescription" IsForward="false">ns=1;i=5002</Reference>
        </References>
        <Value>
            <uax:String>//xs:element[@name='Vector']</uax:String>
        </Value>
    </UAVariable>
    <UAVariable SymbolicName="TypeDictionary_BinarySchema" DataType="ByteString" NodeId="ns=1;i=6001" BrowseName="1:TypeDictionary">
[... 107 lines cut ...]
            <Reference ReferenceType="HasDescription">ns=1;i=6006</Reference>
        </References>
    </UAObject>
</UANodeSet>
`

So, my questions are:
1. How can I solve this issue and successfully compile the server?
2. Once the issue id fixed in the server side, what additional things do I need to do in order to decode and read the contents of myVector from a client?

Thanks in Advance!

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Current version
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-01-15:
Are you sure that you are using the current version?

UA_encodeBinary was completely removed from the nodeset compiler with #2337, but in your code it's still there...

Also, how did you compile the C code from your .xml? Can you post the CMake lines?

MrPoudel, 2019-01-15:
Hi @Pro ,
Thanks for your reply.
Sorry, I am using the old version, it was my mistake. Anyway, my  `../examples/nodeset/CMakeLists.txt` has the following content.
```
# generate namespace from XML file
add_custom_command(OUTPUT ${PROJECT_BINARY_DIR}/src_generated/example_nodeset.c
                   ${PROJECT_BINARY_DIR}/src_generated/example_nodeset.h
                   PRE_BUILD
                   COMMAND ${PYTHON_EXECUTABLE} ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/nodeset_compiler.py
                   --types-array=UA_TYPES
                   --existing ${UA_FILE_NS0}
                   --xml ${PROJECT_SOURCE_DIR}/examples/nodeset/server_nodeset.xml
                   ${PROJECT_BINARY_DIR}/src_generated/example_nodeset
                   DEPENDS ${UA_FILE_NS0}
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/nodeset_compiler.py
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/nodes.py
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/nodeset.py
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/datatypes.py
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/backend_open62541.py
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/backend_open62541_nodes.py
                   ${PROJECT_SOURCE_DIR}/tools/nodeset_compiler/backend_open62541_datatypes.py
                   ${PROJECT_SOURCE_DIR}/examples/nodeset/server_nodeset.xml)

add_example(server_nodeset server_nodeset.c ${PROJECT_BINARY_DIR}/src_generated/example_nodeset.c)

```
In previous version, server_nodeset.xml would be enough to compile a working server. Therefore, I tried with the old version to compile the server, but this time with user defined data types and got the errors. 

However, the new version seems obscure and contains some addition files like server_nodeset.csv (though I can generate it from the UaModeler, I don't know why do we need this, because we already have the NodeId information in the generated xml file). Could you provide some hint  on how can I generate the source files for my information model (I have already provided the content of XML file) and compile successfully?
Thank you very much.

MrPoudel, 2019-01-16:
Hi @matkonnerth,

Thank you very much for your support.
As you suggested, I imported the three files from UaModeler for my information model and renamed as ```testnodeset.xml```, ```testnodeset.csv``` and ```testtypes.bsd``` and I replaced the corresponding files in ```../examples/nodeset/```. 

Then, I successfully compiled the ```server_testnodeset.c``` by setting `UA_NAMESPACE_ZERO` to ` FULL`. 
After that , I wrote a client to read the content of my structure Vector. Following is the code of Client:
``` 
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <ua_client_subscriptions.h>
#include <ua_client_highlevel.h>
#include <ua_config_default.h>
#include <ua_log_stdout.h>

#include "ua_namespace_testnodeset.h"

int main(int argc, char *argv[]) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
     UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }   
    /* Read user defined attribute */     
    printf("\nReading the value of node (2, 6008) :\n");
    UA_Variant val;
    UA_Variant_init(&val);
    retval = UA_Client_readValueAttribute(client, UA_NODEID_NUMERIC(2, 6008), &val);     
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        printf("Error occured!");
        return (int)retval;
    }  
    UA_Vector *v = (UA_Vector*) val.data;    
    printf("Values of vector: X = %lf Y = %lf  Z = %lf \n", v->x, v->y, v->z);   
    
   // UA_Variant_delete(&val);   
    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return (int) UA_STATUSCODE_GOOD;
}

```
Auto generated UA type for Vector is:
```
/**
 * Vector
 * ^^^^^^
 */
typedef struct {
    UA_Double x;
    UA_Double y;
    UA_Double z;
} UA_Vector;

```
I successfully compiled the above client code too, but I was not able to read the content of Vector type node  ,,myVector" correctly. What might be the problem? Am I missing something on client side?
Thanks!

matkonnerth, 2019-01-16:
Hi @MrPoudel ,
take a look on the examples/custom_datatype/client_types_custom.c -  there the customDataTypes are "registered" with 

` UA_ClientConfig config = UA_ClientConfig_default;

    /* Make your custom datatype known to the stack */
    UA_DataType types[1];
    types[0] = PointType;

    /* Attention! Here the custom datatypes are allocated on the stack. So they
     * cannot be accessed from parallel (worker) threads. */
    UA_DataTypeArray customDataTypes = {config.customDataTypes, 1, types};
    config.customDataTypes = &customDataTypes;

    UA_Client *client = UA_Client_new(config);`

Otherwise the stack will have no chance to decode the extension object. 

To the question regarding the encoded type dictionary:
As it is my understanding, the open client will not use the type dictionary for decoding an extension object. The nodeset compiler simple generates a variable node for the TypeDictionary_BinarySchema, allocates memory and sets the value to encoded bytestring.

MrPoudel, 2019-01-18:
Hi @matkonnerth,  

Thanks for your hint.
Now, I am able to read the content of Vector type node ,,myVector" correctly from a client.
One last question, referring to the attached image,

[image]

I removed the content shown in the red box above and compiled the `server_testnodeset.c`, everything worked fine as before. Therefore, I am wondering whether the nodeset compiler has been using the value of this bytestring during the generation of the souce code? If yes, where does exactly this value has influence in the generated code? if not, why the UaModeler is generating this bytestring? 
Thanks!

matkonnerth, 2019-01-21:
Hi @MrPoudel ,
the data in the red box is the TypeDictionary, the nodesetcompiler doesn't interpret it or use it for nodeset generation. He simple adds a variable node and sets the value to this bytestring. The type dictionary is useful for generic clients. For further information, please take a look on this thread: https://github.com/open62541/open62541/issues/1420

MrPoudel, 2019-01-21:
Thank you very much.
I appreciate it!

closed 2019-01-21 by MrPoudel